ni/nidaqmx-python #16: Counter Resetting Every Read

jcox-nd, 2017-08-01:
I am using a USB-6009 multifunction DAQ with Windows 17 and Python 3.6 as a digital counter on the PFI0 port. The counter tests good in NI MAX. Also, if I read multiple samples, the counter does increment as expected when presented with a 1 kHz square wave. 

However, I find that the counter state is reset before every read operation. In other words, if I read 1 sample at a time, I will get the value "0" or "1" as an output. If I read, for example, 1000 samples at once, each sample (more or less, depending on sampling rate) is incrementing as expected. 

I am using the example code in ci_count_edges.py, which uses i_channels.add_ci_count_edges_chan().

The underlying NI DAQ MX function is DAQmxReadCounterU32Ex(), which isn't supposed to do this. Therefore, the cause is a bit of a mystery to me.

epage, 2017-08-17:
Have you tried explicitly starting/stopping the task?

It doesn't look like the example does that but instead [shows the different styles of reading](https://github.com/ni/nidaqmx-python/blob/master/nidaqmx_examples/ci_count_edges.py).  *The example probably needs some work to make it more focused on how someone would want to count edges rather than being an API showcase.*

Why is this needed?  To allow people to optimize their applications, [DAQmx implements a state machine](https://zone.ni.com/reference/en-XX/help/370466AD-01/mxcncpts/taskstatemodel/) but it allows implicit state transitions for ease of use.  Most of the time, when an implicit transition happens, the task automatically rolls back after the call.  In this case I wonder if the task is automatically stopping after the read call because it automatically started.

epage, 2017-09-11:
I created #31 for updating all of the examples.

closed 2017-12-14 by brianlee1200