ni/nidaqmx-python #17: problem with reading from encoders

yotamish, 2017-08-13:
Hi,
I'm using NI6321 with incremental angular encoder.
I use the command: "task1.ci_channels.add_ci_ang_encoder_chan("Dev1/ctr1") "
when I try to read (task1.read function) the angular, I get a steady value of 0.0 even though the correct values are different (as I can see in NIMAX for example)
Do you have an idea why this is happening?
thanks in advance,
Yotam

epage, 2017-08-17:
Can you isolate the problem to code that you can post?

Also, could you provide some additional context for us on this?

- OS: *NAME* *VERSION*
- Python version: *VERSION*
- NI-DAQmx version: *VERSION*

Note: to get the Python version, you can run
```bash
$ python -c "import sys; print(sys.version)"
```

### Installed packages and version

*TODO:* paste the output from
```bash
$ python -m pip list
```

yotamish, 2017-08-17:
hey, thanks for the response.

the problem we are facing is that while using the method add_ci_ang_encoder_chan("Dev1/ctr0") in order to read our angular encoder, we are not getting good readings of the angle - they are not consistent with the correct angles (as can be seen in Matlab for example). what we would like to do is read the angle in some desired frequency (which is less important to us) but we think we are reading "old" angles from the buffer - and we cannot find a way to read angles in real time (like we can do in Matlab or using the interface of the NI card).

OS: Windows7

Python version: 2.7

Ni-DaQmx version: the latest version (I am not at the university at the moment, but I personaly downloaded it last week. I think it was 17.01 or something close)

thank you in advance,
Yotam.
________________________________
מאת: Ed Page <notifications@github.com>
‏‏נשלח: יום חמישי 17 אוגוסט 2017 13:34
‏‏אל: ni/nidaqmx-python
עותק: yotamish; Author
‏‏נושא: Re: [ni/nidaqmx-python] problem with reading from encoders (#17)

Can you isolate the problem to code that you can post?

Also, could you provide some additional context for us on this?

  *   OS: NAME VERSION
  *   Python version: VERSION
  *   NI-DAQmx version: VERSION

Note: to get the Python version, you can run

$ python -c "import sys; print(sys.version)"

Installed packages and version

TODO: paste the output from

$ python -m pip list

epage, 2017-08-29:
To make sure we are understanding your problem correctly, could you please include a simple program that reproduces your problem and what hardware you are using it with?

GlitchModulator, 2017-12-06:
@epage I am also experiencing this problem. I think i have found why (at least in part) the CI channels with type encoder do not return values other than zero.

I am using USB-6210 with a 3 channel encoder (A, B and X channels) on Windows 7. Python3, nidaqmx version 0.5.6

Code:

import nidaqmx
from nidaqmx.constants import AcquisitionType, TaskMode, CountDirection, Edge
from nidaqmx._task_modules.channels.ci_channel import CIChannel
from nidaqmx._task_modules.channel_collection import ChannelCollection
from nidaqmx.constants import AngleUnits
from nidaqmx.constants import EncoderType
 
task = nidaqmx.Task()
#create virtual channel
x = task.ci_channels.add_ci_ang_encoder_chan(counter = 'Dev2/ctr0', decoding_type = EncoderType.X_2, units=AngleUnits.DEGREES, pulses_per_rev=512)
# task.read()

>>> x.chan_type
<ChannelType.COUNTER_INPUT: 10131>
>>> x.ci_meas_type
<UsageTypeCI.POSITION_ANGULAR_ENCODER: 10360>

When i checked chan_type and ci_meas_type, both are set correctly. However in the read() function in Task.py, there is no special case for "UsageTypeCI.POSITION_ANGULAR_ENCODER" (code snippet below). Meaning this defaults to using the read function _read_counter_f_64_ex() (from read_functions.py). Can you shed light on this function? Should this have a different read function for the encoder type?

.... (SNIPPET FROM task.py) starts at line 644 ...
        elif read_chan_type == ChannelType.COUNTER_INPUT:
            meas_type = channels_to_read.ci_meas_type

            if meas_type == UsageTypeCI.PULSE_FREQ:
                frequencies = numpy.zeros(array_shape, dtype=numpy.float64)
                duty_cycles = numpy.zeros(array_shape, dtype=numpy.float64)

                samples_read = _read_ctr_freq(
                    self._handle, frequencies, duty_cycles,
                    number_of_samples_per_channel, timeout)

                data = []
                for f, d in zip(frequencies, duty_cycles):
                    data.append(CtrFreq(freq=f, duty_cycle=d))

            elif meas_type == UsageTypeCI.PULSE_TIME:
                high_times = numpy.zeros(array_shape, dtype=numpy.float64)
                low_times = numpy.zeros(array_shape, dtype=numpy.float64)

                samples_read = _read_ctr_time(
                    self._handle, high_times, low_times,
                    number_of_samples_per_channel, timeout)
                data = []
                for h, l in zip(high_times, low_times):
                    data.append(CtrTime(high_time=h, low_time=l))

            elif meas_type == UsageTypeCI.PULSE_TICKS:
                high_ticks = numpy.zeros(array_shape, dtype=numpy.uint32)
                low_ticks = numpy.zeros(array_shape, dtype=numpy.uint32)

                samples_read = _read_ctr_ticks(
                    self._handle, high_ticks, low_ticks,
                    number_of_samples_per_channel, timeout)
                data = []
                for h, l in zip(high_ticks, low_ticks):
                    data.append(CtrTick(high_tick=h, low_tick=l))

            elif meas_type == UsageTypeCI.COUNT_EDGES:
                data = numpy.zeros(array_shape, dtype=numpy.uint32)

                samples_read = _read_counter_u_32_ex(
                    self._handle, data, number_of_samples_per_channel, timeout)

            else:
                data = numpy.zeros(array_shape, dtype=numpy.float64)

                samples_read = _read_counter_f_64_ex(
                    self._handle, data, number_of_samples_per_channel, timeout)
.... ends at line 693 ...

brianlee1200, 2017-12-08:
Hi @GlitchModulator,

That behavior is correct. The Angular Encoder measurement type return values in the form of doubles, which necessitates the use of `_read_counter_f_64_ex`. 

Are you sure you're setting up your task correctly? For instance, have you configured the A, B and Z input terminals using properties such as [this](http://nidaqmx-python.readthedocs.io/en/latest/ci_channel.html#nidaqmx._task_modules.channels.ci_channel.CIChannel.ci_encoder_a_input_term)? Are you using Implicit timing instead of Sample Clock timing? The behavior of counter tasks configured to have sample clocked timing is complicated; most of our customers actually just need implicit timing for their applications.

The following links may prove helpful for you:
http://www.ni.com/tutorial/7109/en/
https://forums.ni.com/t5/NI-ELVIS-II-and-myDAQ/Angular-Position-Measurement-with-a-Quadrature-Encoder/ta-p/3496226

The `nidaqmx` package is officially supported by NI. For support for your application, you can open a request through the NI support portal at [ni.com](http://www.ni.com/en-us/support.html), where trained Application Engineers can assist you further with your application.

If you believe that this is truly a bug in the Python API, please don't hesitate to contact us again.

GlitchModulator, 2017-12-18:
@brianlee1200 Thank you for the response. I will submit a ticket with NI support and see if that yields anything, however I think my issue is with using the Python API to configure the task correct. I am able to test the counter functionality through NI's Test panels app and it works as expected.

I am setting the A B and X channels correctly, however I had not considered the implicit/sample clock timing. What I really need is an example of how to set up a counter/encoder task in Python - do you know of any example code that i could look at?

Thanks for your help

FrancoisCoteIng, 2018-02-21:
Hi! Did you succeed to read the correct value of your encoder? I still have some issues with mine.

neilvana, 2018-03-06:
This could really use an example since it can be a little more complicated to configure.

SureshVakati, 2019-05-13:
Hi, I am having hard-time to use this class.  How did you set A, B Terminals to read Angular position? 
`task.ci_channels.add_ci_ang_encoder_chan(counter` = 'Dev2/ctr0', decoding_type = EncoderType.X_2, units=AngleUnits.DEGREES, pulses_per_rev=512)`

The above class doesn't take any arguments for terminal numbers. Can anyone help me, please? Thanks.

shingo-shimoda, 2020-08-04:
Finally, I found the method to read encoder in correct way.
We can get the angle from the encoder. Key point is "task.start()"

---------------------------------
with nidaqmx.Task() as task:
        task.ci_channels.add_ci_ang_encoder_chan(counter = 'Dev4/ctr0', decoding_type = EncoderType.X_4, units=AngleUnits.DEGREES, pulses_per_rev=1024, initial_angle=0.0)
        task.start()
        while True:
            ang=task.read() 
--------------------------------

zhindes, 2022-02-08:
This is a general NI-DAQmx problem and not an issue with the `nidaqmx` Python bindings. Consider posting on https://forums.ni.com/ for help resolving issues such as this.

closed 2022-02-08 by zhindes