ni/nidaqmx-python #23: Support querying device calibration information and Self-Cal

epage, 2017-08-18:
I don't think this stuff is supported in the API.

See http://zone.ni.com/reference/en-XX/help/370471AA-01/mxcprop/attributeclasscalibrationinfo/

spalatofhi, 2020-05-26:
Has self calibration (`DAQmxSelfCal`) been added to the API?

spalatofhi, 2020-05-26:
This seems to work, for self-calibration at least. It doesn't raise an error on a simulated device...

```
def self_cal(device: str):
    cfunc = lib_importer.windll.DAQmxSelfCal
    if cfunc.argtypes is None:
        with cfunc.arglock:
            cfunc.argtypes = [ctypes_byte_str]
    retcode = cfunc(device)
    check_for_error(retcode)
    return True # success!
```

flef, 2023-11-07:
Hey, 

Can you propose a Pull Request with this change ? I'm also looking for DAQmxSelfCal function in the API.

Regards

zhindes, 2023-11-07:
@flef - this is a 6-year old Feature request. Nobody here will create a PR with that change. You can if you'd like!

bkeryan, 2023-12-11:
Updated link to calibration info properties: https://www.ni.com/docs/en-US/bundle/ni-daqmx-c-api-ref/page/mxcprop/attributeclasscalibrationinfo.html