ni/nidaqmx-python #30: callback_data parameter missing from event_registration functions

hannoeichel, 2017-09-04:
For example, the function
` def register_every_n_samples_acquired_into_buffer_event(self, sample_interval, callback_method):`
has no `callback_data` as a parameter, but the documentation describes:
The callback_data parameter contains the value you passed in the callback_data parameter of this function.
Is it not implemented yet? Or am I missing something?

brianlee1200, 2017-09-05:
We intentionally simplified the callback function signature to omit the `callback_method` parameter since we've determined there are alternate ways of passing data between callback functions and the main context. 

Check out the following example:
https://github.com/ni/nidaqmx-python/blob/master/nidaqmx_examples/every_n_samples_event.py

closed 2017-09-05 by brianlee1200

zerkerX, 2022-02-02:
That is not the case if you are trying to pass data back from a task to a Python class instance.