ni/nidaqmx-python #33: Error when I call python APIs

ardeal, 2017-09-19:
On Windows 10, I installed anaconda 2.7 and PyCharm to acquire usb-6009 data.
the driver of usb6009 is correctly installed as I can observe data from "Measurement & Automation Explorer" correctly.

import nidaqmx

with nidaqmx.Task() as task:
    aaa=0
    task.ai_channels.add_ai_voltage_chan("Dev1/ai0")
    print task.read()

when I run the upper code, I got the following error:

Traceback (most recent call last):
  File "C:\Program Files\JetBrains\PyCharm Community Edition 2017.1.5\helpers\pydev\pydevd.py", line 1591, in <module>
    globals = debugger.run(setup['file'], None, None, is_module)
  File "C:\Program Files\JetBrains\PyCharm Community Edition 2017.1.5\helpers\pydev\pydevd.py", line 1018, in run
    pydev_imports.execfile(file, globals, locals)  # execute the script
  File "C:/work/code_python/mwsignal/nidaq_mw.py", line 23, in <module>
    with nidaqmx.Task() as task:
  File "C:\ProgramData\Anaconda2\lib\site-packages\nidaqmx\task.py", line 84, in __init__
    self._handle = lib_importer.task_handle(0)
  File "C:\ProgramData\Anaconda2\lib\site-packages\nidaqmx\_lib.py", line 178, in task_handle
    self._parse_typedefs()
  File "C:\ProgramData\Anaconda2\lib\site-packages\nidaqmx\_lib.py", line 235, in _parse_typedefs
    version = system.driver_version
  File "C:\ProgramData\Anaconda2\lib\site-packages\nidaqmx\system\system.py", line 73, in driver_version
    return DriverVersion(self._major_version, self._minor_version,
  File "C:\ProgramData\Anaconda2\lib\site-packages\nidaqmx\system\system.py", line 108, in _major_version
    cfunc = lib_importer.windll.DAQmxGetSysNIDAQMajorVersion
  File "C:\ProgramData\Anaconda2\lib\site-packages\nidaqmx\_lib.py", line 166, in windll
    self._import_lib()
  File "C:\ProgramData\Anaconda2\lib\site-packages\nidaqmx\_lib.py", line 203, in _import_lib
    windll = ctypes.windll.LoadLibrary(lib_name)
  File "C:\ProgramData\Anaconda2\lib\ctypes\__init__.py", line 440, in LoadLibrary
    return self._dlltype(name)
  File "C:\ProgramData\Anaconda2\lib\ctypes\__init__.py", line 362, in __init__
    self._handle = _dlopen(self._name, mode)
TypeError: LoadLibrary() argument 1 must be string, not unicode
Exception AttributeError: "'Task' object has no attribute '_handle'" in <object repr() failed> ignored

Process finished with exit code 1

brianlee1200, 2017-09-19:
Which version of Python are you using? This looks to be a [known issue](https://github.com/ni/nidaqmx-python/issues/1) in Python 2.7.13 where _ctypes.LoadLibrary no longer accepts Unicode objects.

The current recommendation would for you to downgrade your version of Python to 2.7.12 to get **nidaqmx** working.

ardeal, 2017-09-20:
I am using "Python 2.7.13 :: Anaconda 4.4.0 (64-bit)"

I am wondering why you don't fix the known issue on the latest python.

ardeal, 2017-09-20:
I have just now re-installed anaconda 4.2.0 in which the version of python is 2.7.12.
I add the following code to print the version of python used:
import platform

print platform.python_version()
the printed versionis: 2.7.12

I can acquire data from usb-6009 in python code. if I add the following code, there is error:

with nidaqmx.Task() as task:
    aaa=0
    mwchan = task.ai_channels.add_ai_voltage_chan("Dev1/ai0") 
    print mwchan.ai_ac_excit_freq
    plt.figure()

2.7.12
Backend Qt5Agg is interactive backend. Turning interactive mode on.
Traceback (most recent call last):
  File "C:\Program Files\JetBrains\PyCharm Community Edition 2017.1.5\helpers\pydev\pydevd.py", line 1591, in <module>
    globals = debugger.run(setup['file'], None, None, is_module)
  File "C:\Program Files\JetBrains\PyCharm Community Edition 2017.1.5\helpers\pydev\pydevd.py", line 1018, in run
    pydev_imports.execfile(file, globals, locals)  # execute the script
  File "C:/work/code_python/mwsignal/nidaq_mw.py", line 109, in <module>
    print mwchan.ai_ac_excit_freq
  File "C:\Program Files\Anaconda2\lib\site-packages\nidaqmx\_task_modules\channels\ai_channel.py", line 62, in ai_ac_excit_freq
    check_for_error(error_code)
  File "C:\Program Files\Anaconda2\lib\site-packages\nidaqmx\errors.py", line 127, in check_for_error
    raise DaqError(error_buffer.value.decode("utf-8"), error_code)
nidaqmx.errors.DaqError: Specified property is not supported by the device or is not applicable to the task.
Property: DAQmx_AI_ACExcit_Freq
Channel Name: Dev1/ai0

Task Name: _unnamedTask<0>

Status Code: -200452

ardeal, 2017-09-20:
if I use the following code to set the freq:
mwchan.ai_ac_excit_freq = 1000.0

I still got the following error:
2.7.12
Backend Qt5Agg is interactive backend. Turning interactive mode on.
Traceback (most recent call last):
  File "C:\Program Files\JetBrains\PyCharm Community Edition 2017.1.5\helpers\pydev\pydevd.py", line 1591, in <module>
    globals = debugger.run(setup['file'], None, None, is_module)
  File "C:\Program Files\JetBrains\PyCharm Community Edition 2017.1.5\helpers\pydev\pydevd.py", line 1018, in run
    pydev_imports.execfile(file, globals, locals)  # execute the script
  File "C:/work/code_python/mwsignal/nidaq_mw.py", line 110, in <module>
    mwchan.ai_ac_excit_freq = 1000.0
  File "C:\Program Files\Anaconda2\lib\site-packages\nidaqmx\_task_modules\channels\ai_channel.py", line 78, in ai_ac_excit_freq
    check_for_error(error_code)
  File "C:\Program Files\Anaconda2\lib\site-packages\nidaqmx\errors.py", line 127, in check_for_error
    raise DaqError(error_buffer.value.decode("utf-8"), error_code)
nidaqmx.errors.DaqError: Specified property is not supported by the device or is not applicable to the task.
Property: DAQmx_AI_ACExcit_Freq
Channel Name: Dev1/ai0

Task Name: _unnamedTask<0>

Status Code: -200452

epage, 2017-09-20:
So for this block:
```python
with nidaqmx.Task() as task:
    aaa=0
    mwchan = task.ai_channels.add_ai_voltage_chan("Dev1/ai0")
    print mwchan.ai_ac_excit_freq()
    plt.figure()
```
you are getting this error

> nidaqmx.errors.DaqError: Specified property is not supported by the device or is not applicable to the task.

When a DAQmx property is supported, it comes with a default value and does not need to be set before reading it.

What the error message is trying to say is that either the device doesn't support AC Excitation Frequency or the type of task or channel doesn't support it.  For example, you can't set thermocouple properties on a channel created with `add_ai_voltage_chan`.  Instead you need to add a thermocouple channel.

In this case, I believe its because the USB-6009 doesn't support `.ai_ac_excit_freq` on its channels.  While our python documentation doesn't list what properties are supported on what types of hardware, our [C documentation does](http://zone.ni.com/reference/en-XX/help/370471AA-01/TOC40.htm) and in the area of `Analog Input :: General Properties :: Signal Conditioning :: Excitation :: AC Excitation :: Frequency`, I do not see it for the [USB-6009](http://zone.ni.com/reference/en-XX/help/370471AA-01/cdaqmxsupp/usb-6009/) but [SCXI-1540](http://zone.ni.com/reference/en-XX/help/370471AA-01/cdaqmxsupp/scxi-1540/) is an example of where it shows up.

Some side notes
- If the property was supported, you would have run into another issue with `print mwchan.ai_ac_excit_freq()`.  `ai_excit_freq` is a property, not a function, so it should be `print mwchan.ai_ac_excit_freq` to access the value.
- It is usually best to have one github Issue per problem unless you feel they might be related so that its easier to follow the conversation, other people can find related problems, etc.
- While I suggested the discussion forum for #34, I feel like this is a different case because it can be unclear whether an error is a bug in DAQmx or not.

closed 2017-12-14 by brianlee1200