ni/nidaqmx-python #34: how to configure usb-6009 from Python

ardeal, 2017-09-20:
I would like to configure the USB-6009 as the following parameters:
1) sample rate 1000Hz
2) acquistion mode: continuous
3) input configuration: RSE

I can read usb-6009 by the following code, but I need to re-configure the 6009 by those upper parameters.
with nidaqmx.Task() as task:
    task.ai_channels.add_ai_voltage_chan("Dev1/ai0")
    f = open('data_ni_noise_3.csv', 'w+')
    f.write('data0,timestamp\n')
    while True:
        onedata = task.read()
        f.write(str(onedata)+',0\n')
    f.close()

Thanks!

ardeal, 2017-09-20:
I use the following Python code to read data from usb6009:
plt.figure()
with nidaqmx.Task() as task:
    mwchan = task.ai_channels.add_ai_voltage_chan("Dev1/ai0")
    # task.start()
    # print task.timing
    while True:
        onedata = task.read()
        datalist.append(onedata)
        datalist.pop(0)
        plt.clf()
        plt.plot(datalist)
        plt.show(0)
        plt.pause(0.0000001)

The data can be correctly read from usb6009, but the data refresh speed is very low.
it seems that onedata = task.read() statements wates much time.

epage, 2017-09-20:
Since this is product support, you might get better help if you go to the [discussion forum](https://forums.ni.com/t5/Multifunction-DAQ/bd-p/250) but I do feel this shows a lacking in the examples / documentation (see #16 which led to #31).

To get you started in the right direction, you'll want to [configure timing](https://nidaqmx-python.readthedocs.io/en/latest/timing.html#nidaqmx._task_modules.timing.Timing.cfg_samp_clk_timing) and read blocks of data at a time via [`number_of_samples_per_channel`](https://nidaqmx-python.readthedocs.io/en/latest/task.html#nidaqmx.task.Task.read).

Tip: when posting code, if you start your post with 3 back-ticks and end it with three back-ticks, it'll retain the formatting (like indentation) and syntax highlight it.  See the [code section](https://guides.github.com/features/mastering-markdown/).

Example:
```python
with open('data_ni_noise_3.csv', 'w+') as f:
    f.write('data0,timestamp\n')
    with nidaqmx.Task() as task:
        task.ai_channels.add_ai_voltage_chan("Dev1/ai0")
        while True:
            onedata = task.read()
            f.write('{},0\n'.format(onedata))
```

ardeal, 2017-09-21:
Thanks!

Do you have the example code to read back the data from usb6009 quickly?
at present, the only issue for me is that I can read data but not quickly.

ardeal, 2017-09-21:
Let me try back-ticks you mentioned:

'''
    for i in range(0, 2):
        #for i in range(0, 10000):
        # onedata = task.read()
        data = task.read(number_of_samples_per_channel=10240)
        # datalist.append(onedata)
        # datalist.pop(0)

        ax.clear()
        ax.plot(data)
        fig.show(0)
        plt.pause(0.000000001)
'''

ardeal, 2017-09-21:
'''
    for i in range(0, 2):
        #for i in range(0, 10000):
        # onedata = task.read()
        data = task.read(number_of_samples_per_channel=10240)
        # datalist.append(onedata)
        # datalist.pop(0)

        ax.clear()
        ax.plot(data)
        fig.show(0)
        plt.pause(0.000000001)
'''

ardeal, 2017-09-21:
'''    t1 = datetime.utcnow()
    # while True:
    for i in range(0, 2):
        #for i in range(0, 10000):
        # onedata = task.read()
        data = task.read(number_of_samples_per_channel=10240)
        # datalist.append(onedata)
        # datalist.pop(0)
        ax.clear()
        ax.plot(data)
        fig.show(0)
        plt.pause(0.000000001)'''

ardeal, 2017-09-21:
'''
    t1 = datetime.utcnow()
    # while True:
    for i in range(0, 2):
        #for i in range(0, 10000):
        # onedata = task.read()
        data = task.read(number_of_samples_per_channel=10240)
        # datalist.append(onedata)
        # datalist.pop(0)
        ax.clear()
        ax.plot(data)
        fig.show(0)
        plt.pause(0.000000001)
'''

ardeal, 2017-09-21:
'''
    t1 = datetime.utcnow()
    # while True:
    for i in range(0, 2):
        #for i in range(0, 10000):
        # onedata = task.read()
        data = task.read(number_of_samples_per_channel=10240)
        # datalist.append(onedata)
        # datalist.pop(0)
        ax.clear()
        ax.plot(data)
        fig.show(0)
        plt.pause(0.000000001)
        # task.
        # strinput = raw_input('')
        # if strinput =='p':
        #     break
        # f.write(str(onedata)+',0\n')
    f.close()
    t2 = datetime.utcnow()
    print (t2-t1).microseconds
'''

ardeal, 2017-09-21:
'''javascript
    t1 = datetime.utcnow()
    # while True:
    for i in range(0, 2):
        #for i in range(0, 10000):
        # onedata = task.read()
        data = task.read(number_of_samples_per_channel=10240)
        # datalist.append(onedata)
        # datalist.pop(0)
        ax.clear()
        ax.plot(data)
        fig.show(0)
        plt.pause(0.000000001)
        # task.
        # strinput = raw_input('')
        # if strinput =='p':
        #     break
        # f.write(str(onedata)+',0\n')
    f.close()
    t2 = datetime.utcnow()
    print (t2-t1).microseconds
'''

ardeal, 2017-09-21:
Let me try 3 backticks:

'''
    t1 = datetime.utcnow()
    # while True:
    for i in range(0, 2):
        #for i in range(0, 10000):
        # onedata = task.read()
        data = task.read(number_of_samples_per_channel=10240)
        # datalist.append(onedata)
        # datalist.pop(0)
        ax.clear()
        ax.plot(data)
        fig.show(0)
        plt.pause(0.000000001)
        # task.
        # strinput = raw_input('')
        # if strinput =='p':
        #     break
        # f.write(str(onedata)+',0\n')
    f.close()
    t2 = datetime.utcnow()
    print (t2-t1).microseconds
'''

ardeal, 2017-09-21:
`    t1 = datetime.utcnow()
    # while True:
    for i in range(0, 2):
        #for i in range(0, 10000):
        # onedata = task.read()
        data = task.read(number_of_samples_per_channel=10240)
        # datalist.append(onedata)
        # datalist.pop(0)
        ax.clear()
        ax.plot(data)
        fig.show(0)
        plt.pause(0.000000001)
        # task.
        # strinput = raw_input('')
        # if strinput =='p':
        #     break
        # f.write(str(onedata)+',0\n')
    f.close()
    t2 = datetime.utcnow()
    print (t2-t1).microseconds`
 t1 = datetime.utcnow()
    # while True:
    for i in range(0, 2):
        #for i in range(0, 10000):
        # onedata = task.read()
        data = task.read(number_of_samples_per_channel=10240)
        # datalist.append(onedata)
        # datalist.pop(0)
        ax.clear()
        ax.plot(data)
        fig.show(0)
        plt.pause(0.000000001)
        # task.
        # strinput = raw_input('')
        # if strinput =='p':
        #     break
        # f.write(str(onedata)+',0\n')
    f.close()
    t2 = datetime.utcnow()
    print (t2-t1).microseconds

ardeal, 2017-09-21:
`
    t1 = datetime.utcnow()
    # while True:
    for i in range(0, 2):
        #for i in range(0, 10000):
        # onedata = task.read()
        data = task.read(number_of_samples_per_channel=10240)
        # datalist.append(onedata)
        # datalist.pop(0)
        ax.clear()
        ax.plot(data)
        fig.show(0)
        plt.pause(0.000000001)
        # task.
        # strinput = raw_input('')
        # if strinput =='p':
        #     break
        # f.write(str(onedata)+',0\n')
    f.close()
    t2 = datetime.utcnow()
    print (t2-t1).microseconds

`

ardeal, 2017-09-21:
`    t1 = datetime.utcnow()
    # while True:
    for i in range(0, 2):
        #for i in range(0, 10000):
        # onedata = task.read()
        data = task.read(number_of_samples_per_channel=10240)
        # datalist.append(onedata)
        # datalist.pop(0)
        ax.clear()
        ax.plot(data)
        fig.show(0)
        plt.pause(0.000000001)
        # task.
        # strinput = raw_input('')
        # if strinput =='p':
        #     break
        # f.write(str(onedata)+',0\n')
    f.close()
    t2 = datetime.utcnow()
    print (t2-t1).microseconds`

ardeal, 2017-09-21:
```

with nidaqmx.Task() as task:
    mwchan = task.ai_channels.add_ai_voltage_chan("Dev1/ai0")
    task.timing.cfg_samp_clk_timing(fs)
    # print mwchan.physical_channel.ai_term_cfgs

    while True:
        tdata = task.read(number_of_samples_per_channel=nfft)
        # data_all_list.append(tdata)
        # testdata = daq.read_usb6009(fs, nfft)
        testdata = np.array(tdata)
        testdata -= np.mean(testdata)

        ffto = np.fft.fft(testdata) / nfft
        ffto = np.abs(ffto)
        ffto = ffto[0:(nfft / 2 + 1)]
        ffto[0] = 0
        bf = bis.bispectrum(testdata, nfft)
        fdata = bf[0:(nfft / 2 + 1)]
        fdata[0] = 0

```

ardeal, 2017-09-21:
I can read usb-6009 data quickly by the following code:

I would like to configure usb-6009 

acquistion mode: continuous
input configuration: RSE(Referrenced Single Ended)

The following command can configure sample rate correctly.
```
task.timing.cfg_samp_clk_timing(fs)
```
Can you tell me the command to configure it for acquisition mode and input configuration?

```
data_all_list = []

with nidaqmx.Task() as task:
    mwchan = task.ai_channels.add_ai_voltage_chan("Dev1/ai0")
    task.timing.cfg_samp_clk_timing(fs)
    # print mwchan.physical_channel.ai_term_cfgs

    # while True:
    for i in range(0, 200):
        tdata = task.read(number_of_samples_per_channel=nfft)
        data_all_list += tdata
```

epage, 2017-09-21:
I currently don't have a system setup for this, so from reading the documentation, my guess is

```python
with nidaqmx.Task() as task:
    mwchan = task.ai_channels.add_ai_voltage_chan("Dev1/ai0", terminal_config=nidaqmx.TerminalConfiguration.RSE)
    task.timing.cfg_samp_clk_timing(fs, sample_mode=nidaqmx.AcquisitionType.CONTINUOUS)
    # print mwchan.physical_channel.ai_term_cfgs

    # while True:
    for i in range(0, 200):
        tdata = task.read(number_of_samples_per_channel=nfft)
        data_all_list += tdata
```

Relevant parts of the docs
- For setting RSE
  - [add_ai_voltage_chan](https://nidaqmx-python.readthedocs.io/en/latest/ai_channel_collection.html#nidaqmx._task_modules.ai_channel_collection.AIChannelCollection.add_ai_voltage_chan)
  - [TerminalConfiguration enum](https://nidaqmx-python.readthedocs.io/en/latest/constants.html#nidaqmx.constants.TerminalConfiguration)
  - [ai_term_cfg property](https://nidaqmx-python.readthedocs.io/en/latest/ai_channel.html#nidaqmx._task_modules.channels.ai_channel.AIChannel.ai_term_cfg) to see what you're task is configured for and/or to change it dynamically
- For setting continuous
  - [cfg_samp_clk_timing](https://nidaqmx-python.readthedocs.io/en/latest/timing.html#nidaqmx._task_modules.timing.Timing.cfg_samp_clk_timing)
  - [samp_quant_samp_mode property](https://nidaqmx-python.readthedocs.io/en/latest/timing.html#nidaqmx._task_modules.timing.Timing.samp_quant_samp_mode) to see what you're task is configured for and/or to change it dynamically.

ardeal, 2017-09-22:
```python

fig = plt.figure()
ax1 = fig.add_subplot(211)
ax2 = fig.add_subplot(212)

nfft = 1024
fs = 1000.0
freqs = np.linspace(0, fs/2, nfft/2+1)

with nidaqmx.Task() as task:
    task.ai_channels.add_ai_voltage_chan("Dev1/ai0", terminal_config=TerminalConfiguration.RSE, min_val=-5.0, max_val=5.0, units=VoltageUnits.VOLTS)
    task.timing.cfg_samp_clk_timing(fs, active_edge=Edge.RISING, sample_mode=AcquisitionType.CONTINUOUS, samps_per_chan=nfft)

    # task.start()
    # task.stop()
    while True:

        t1 = (int(round(time.time() * 1000)))
        data = task.read(number_of_samples_per_channel=nfft)
        t2 = (int(round(time.time() * 1000)))
        tdata = np.array(data)
        tdata -= np.mean(tdata)
        ffto = np.fft.fft(tdata)
        ffto = np.abs(ffto)
        ffto = ffto[0:(nfft/2+1)]

        ax1.clear()
        ax2.clear()
        ax1.plot(data)
        ax2.plot(freqs, ffto)
        fig.show(0)
        plt.pause(0.000000001)
        t3 = (int(round(time.time() * 1000)))
        print 't2-t1 = {}ms, t3-t2 ={}ms'.format(t2-t1, t3-t2)

```

closed 2017-09-22 by ardeal

epage, 2017-09-22:
Glad to see you got it working!

aaronschoettger, 2018-06-29:
Just FYI I had to Referance the full path for the Configuration, and my created Task looked like this -- > 
```
task.ai_channels.add_ai_voltage_chan(channel_val, terminal_config=nidaqmx.constants.TerminalConfiguration.RSE, min_val=-10.0, max_val=10.0)
```

poletts, 2018-10-25:
I spent a lot of time trying to set this think.
The last one worked for me, thank you very much guys.

epage, 2018-10-25:
> The last one worked for me, thank you very much guys.

Huh, I realized we might want to highlight that post, so I've hidden the earlier ones to help get people to it faster.

MuhammadUtoro, 2022-10-28:
> ```python
> fig = plt.figure()
> ax1 = fig.add_subplot(211)
> ax2 = fig.add_subplot(212)
>
> nfft = 1024
> fs = 1000.0
> freqs = np.linspace(0, fs/2, nfft/2+1)
>
> with nidaqmx.Task() as task: [...]

Hi ardeal! I have a question on your script, can we run a liveplotting? because once I tried the live plotting error popped out. It says that "Unable to keep up with the acquisition"
Thank you for your help