ni/nidaqmx-python #35: set sample rate of usb-6009

ardeal, 2017-09-22:
I use the following code to configure usb6009, and the printed time(ms) is:

My intention is set the sample rate as 1000Hz. so to read 1024 sample points should need around 1024ms. the printed t2-t1 is around 770ms. 

if I change sample rate to 2000Hz, t2-t1 is still 770ms.... it seems that setting sample rate doesn't take effect.

```
t2-t1 = 779ms, t3-t2 =266ms
t2-t1 = 758ms, t3-t2 =249ms
t2-t1 = 775ms, t3-t2 =260ms
t2-t1 = 763ms, t3-t2 =250ms
t2-t1 = 774ms, t3-t2 =259ms
t2-t1 = 765ms, t3-t2 =263ms
t2-t1 = 762ms, t3-t2 =256ms
t2-t1 = 768ms, t3-t2 =256ms
t2-t1 = 767ms, t3-t2 =270ms

```

code to configure and acqure data:

```

fig = plt.figure()
ax1 = fig.add_subplot(211)
ax2 = fig.add_subplot(212)

nfft = 1024
fs = 1000.0
freqs = np.linspace(0, fs/2, nfft/2+1)

with nidaqmx.Task() as task:
    task.ai_channels.add_ai_voltage_chan("Dev1/ai0", terminal_config=TerminalConfiguration.RSE, min_val=-5.0, max_val=5.0, units=VoltageUnits.VOLTS)
    task.timing.cfg_samp_clk_timing(fs, active_edge=Edge.RISING, sample_mode=AcquisitionType.CONTINUOUS, samps_per_chan=nfft)

    # task.start()
    # task.stop()
    while True:

        t1 = (int(round(time.time() * 1000)))
        data = task.read(number_of_samples_per_channel=nfft)
        t2 = (int(round(time.time() * 1000)))
        tdata = np.array(data)
        tdata -= np.mean(tdata)
        ffto = np.fft.fft(tdata)
        ffto = np.abs(ffto)
        ffto = ffto[0:(nfft/2+1)]

        ax1.clear()
        ax2.clear()
        ax1.plot(data)
        ax2.plot(freqs, ffto)
        fig.show(0)
        plt.pause(0.000000001)
        t3 = (int(round(time.time() * 1000)))
        print 't2-t1 = {}ms, t3-t2 ={}ms'.format(t2-t1, t3-t2)

```

epage, 2017-09-25:
> if I change sample rate to 2000Hz, t2-t1 is still 770ms.... it seems that setting sample rate doesn't take effect.

DAQmx hardware cannot support an infinite range of rates and the supported rates vary based on your hardware's capabilities.  DAQmx ends up coercing the supplied rate to a supported value.  

You can see this by calling:
```python
print(task.timing.samp_clk_rate)
```

Tip
- You can use [`AnalogSingleChannelReader`](https://nidaqmx-python.readthedocs.io/en/latest/stream_readers.html#nidaqmx.stream_readers.AnalogSingleChannelReader) to read numpy data natively rather than `task.read(...)` which returns python lists.

ardeal, 2017-09-26:
Thanks!

I don't know how to initialize the class:

```python
nidaqmx.stream_readers
```
Do you have exampele code?

epage, 2017-09-26:
You can find some in the [tests](https://github.com/ni/nidaqmx-python/blob/master/nidaqmx/tests/test_stream_analog_readers_writers.py#L112).

ardeal, 2017-09-27:
Thanks!

I eventually got to know how to read data by stream_reader. It looks that there is no difference between stream_reader and task. The only difference might be the type of output data.

```python
with nidaqmx.Task() as read_task:
    read_task.ai_channels.add_ai_voltage_chan("Dev1/ai0", terminal_config=TerminalConfiguration.RSE, min_val=-5.0, max_val=5.0, units=VoltageUnits.VOLTS)
    read_task.timing.cfg_samp_clk_timing(fs, active_edge=Edge.RISING, sample_mode=AcquisitionType.CONTINUOUS, samps_per_chan=nfft)
    reader = AnalogSingleChannelReader(read_task.in_stream)
    values_read = numpy.zeros(nfft, dtype=numpy.float64)
    reader.read_many_sample(values_read, number_of_samples_per_channel=nfft, timeout=2)
```

closed 2017-12-14 by brianlee1200