ni/nidaqmx-python #36: no version information available

dolhop, 2017-10-03:
There used to be a version.py file in nidaqmx, which seems to have been removed.  This was useful in that I could check nidaqmx.version and ensure that the minimum revision was installed for my application. 

Or, please provide a nidaqmx.__version__ like most python packages do.

The fact that older versions had a version.py file means that upgrading will leave the old file there, thus confusing the situation.

epage, 2017-10-11:
Are you looking for the python package version or the driver version?

(I'm assuming package version but wanting to confirm)

dolhop, 2017-10-11:
Yes, python package version.  I see the ability to detect the driver
versions.

Thanks.

PR #134 (2022-01-24): Prepare nidaqmx-python for the first release in 5 years
_(Note: these commits will be preserved and not squashed when the PR is completed.)_

# Major Changes

* Switched to [poetry](https://python-poetry.org/) build system.
* Updated supported Python versions to 3.7, 3.8, 3.9, and 3.10
* Updated to latest dependencies
    * Note: This required replacing `numpy.bool` with `bool` native type to resolve warnings. These are synonymous now.

# Resolved Issues

* Fixed `test_many_sample_pulse_ticks` test which was previously disabled due to a DAQmx bug that was fixed.
* Added a bridge device to test [... cut]
closed 2022-01-24 by zhindes