ni/nidaqmx-python #40: is_task_done() cannot be used

MaxP92, 2017-12-01:
When I do:
`
>>>import nidaqmx
>>>nidaqmx.Task()
>>>task.is_task_done() 
`

I get : 

`
error
"nidaqmx\task.py", line 522, in is_task_done
    return is_task_done.value
nidaqmx\_lib.py", line 35, in _getter
    return bool(ctypes.c_uint.value.fget(self))
AttributeError: 'getset_descriptor' object has no attribute 'fget'
`

------------
OS : windows 7
Python : 3.5.2 with Anaconda 4.2.0 (32bits)
NIDAQmx : print(system.driver_version) = DriverVersion(major_version=9, minor_version=2, update_version=3)

brianlee1200, 2017-12-01:
Hey Max,

Is that the exact code you're trying to execute? It looks like you're missing the assignment to `task`, i.e.:

`task = nidaqmx.Task()`

MaxP92, 2017-12-01:
Hello,

Sorry, wrong typing (see the updated code)

Actually I searched a bit and I found two options:
- using the DAQmxGetTaskComplete property (available?)
- Going in task.py and, in def is_task_done(self), change c_bool32() by ctypes.c_bool()

The last one works

Is there another solution ?

brianlee1200, 2017-12-08:
Hi Max,

Thanks for bringing this issue to our attention. I'll submit a fix for this shortly.

closed 2017-12-14 by brianlee1200