ni/nidaqmx-python #44: task change event with nimi nidcpower sequence sweep

hillbilly3456avk, 2017-12-07:
Hi,
I have the following use case. I define a sequence on a nidcpower device using the nimi nidcpower lib. during this sequence I need to observe if a digital input line changes its state. If the state changes I need to measure the nidcpower voltage.

Problem is, that the implementation of `wait_for_event(nidcpower.Event.SEQUENCE_ENGINE_DONE, 10)` is obviously blocking. Means that my callbacks for the registered events are lost. At least I do not enter the callback at all. So I refactored the implementation to use a thread for both, the nidaqmx task, and nidcpower `wait_for_event(nidcpower.Event.SEQUENCE_ENGINE_DONE, 10)` . Unfortunately the callback is still not invoked. Is this because the sync option in the c_func call for register_signal_event is always '0'? compare this help site
[http://zone.ni.com/reference/en-XX/help/370471AE-01/daqmxcfunc/daqmxregistersignalevent/](url)

If this is the issue, what would be the correct value for this option to be able to invoke the callback from within the thread?

Or is my use case not supported?

hillbilly3456avk, 2017-12-07:
okay - I got it. Would this be something for the examples?
```
import random
import time
from threading import Thread
import threading
import nidaqmx
import nidcpower
from nidaqmx.constants import AcquisitionType

def test_signal_event(x_series_device='PXI1Slot2'):
        DaqDcPowerCh0=nidcpower.Session('PXI1Slot9', '0')
        
        DaqDcPowerCh0.source_mode = nidcpower.SourceMode.SEQUENCE
        DaqDcPowerCh0.output_function = nidcpower.OutputFunction.DC_VOLTAGE
        DaqDcPowerCh0.current_limit_autorange = nidcpower.CurrentLimitAutorange.ON   #self.__actualLine['muxPwr1Current']
        DaqDcPowerCh0.voltage_level_autorange = nidcpower.VoltageLevelAutorange.ON
        DaqDcPowerCh0.source_delay = 0.1 # wait 100ms per sequence step
        voltageSweep=create_sweep(19, 16, 0.1)
        sourcedelay = [i[0] for i in voltageSweep]
        sweep = [i[1] for i in voltageSweep]
        DaqDcPowerCh0.set_sequence(sourcedelay, sweep)
        DaqDcPowerCh0.measure_when = nidcpower.MeasureWhen.ON_DEMAND
        with(DaqDcPowerCh0.initiate()):
            myThOb1=myThread()
            myThOb1.setName('thread1')
            myThOb2=myThread2(DaqDcPowerCh0)
            myThOb2.setName('thread2')
            myThOb1.start()
            myThOb2.start()
            
            myThOb1.join()
            myThOb2.join()
            
        
def create_sweep(begin_value, end_value, step_width):
    sweep = []
    number_of_steps=abs(begin_value-end_value) / step_width
    for i in range(int(number_of_steps)):
        step_size = (end_value - begin_value) / number_of_steps
        sweep.append(([abs(step_size), begin_value + i * step_size]))
    return sweep            
            
            
class myThread(Thread):
    exit_flag = threading.Event()
    myTask = None
    def __init__(self):
        ''' Constructor. '''
        Thread.__init__(self)
        self.myTask=nidaqmx.Task()
        self.myTask.di_channels.add_di_chan('PXI1Slot2/port0/line0')
        
        self.myTask.timing.cfg_change_detection_timing(rising_edge_chan='PXI1Slot2/port0/line0', falling_edge_chan='PXI1Slot2/port0/line0',  sample_mode=AcquisitionType.CONTINUOUS)
        self.myTask.register_signal_event(nidaqmx.constants.Signal.CHANGE_DETECTION_EVENT, self.callback)
        self.myTask.start()
 
    def run(self):
        while not self.exit_flag.wait(timeout=1):
            print('Thread %s waits another second' % (self.getName()))
    
    def callback(self,task_handle=0, every_n_samples_event_type=0,
                         number_of_samples=0, callback_data=0):
        print('callback invoked')
        return 0  
    
class myThread2(Thread):
    def __init__(self, DaqDcPowerCh0):
        ''' Constructor. '''
        Thread.__init__(self)
        self.DaqDcPowerCh0 = DaqDcPowerCh0
 
    def run(self):
        self.DaqDcPowerCh0.wait_for_event(nidcpower.Event.SEQUENCE_ENGINE_DONE, 10)
        print('Thread %s is done' % (self.getName()))
        
        
        
test_signal_event()
```

brianlee1200, 2017-12-08:
Hi,

Thanks for contributing an example. We'll vet the example to see we can't add it into the list of examples that use multiple NI driver APIs.

Keep in mind that `nidcpower` is still under development. If you're experiencing any difficulties with `nidcpower`, do post issues - not here, but on `nidcpower`'s [GitHub issues page](https://github.com/ni/nimi-python/issues).

closed 2017-12-08 by brianlee1200