ni/nidaqmx-python #46: stream_writers.write_one_sample : auto_start arg ?

MaxP92, 2017-12-14:
Hello,

In [read the docs](http://nidaqmx-python.readthedocs.io/en/latest/stream_writers.html) about stream_writers, it says 'write_one_sample' has the optional 'auto_start' argument, which is false

Also the def description in stream_writers.py indicates it

MaxP92, 2017-12-14:
Actually the whole stream_writers.write_one_sample is bad : it requires a numpy array even if one element has to be written

Also with 2 channels AO, it raises an error 'No. of dimensions of NumPy Array required: 1' when passing a (2,1) data array, which is yet the correct numpy shape for 2 channels
(with a (1,1) array, it raises the error DAQmxErrors.WRITE_NUM_CHANS_MISMATCH)

brianlee1200, 2017-12-14:
Hi Max,

For the `auto_start` parameter, it does not say the default value is `False`. The docs specify:

> If you do not specify a value for this parameter, NI-DAQmx determines its value based on the type of write method used. If you use a one sample write method, its value is True; conversely, if you use a many sample write method, its value is False.

Please elaborate on your second problem, or post an example code. The `write_one_sample` method for a multi channel writer will always require a NumPy array, even if there is only one channel in the task. Have you tried using a single channel writer?

There are some examples/unit tests here for your reference:
https://github.com/ni/nidaqmx-python/blob/master/nidaqmx/tests/test_stream_analog_readers_writers.py
https://github.com/ni/nidaqmx-python/blob/master/nidaqmx/tests/test_stream_counter_readers_writers.py
https://github.com/ni/nidaqmx-python/blob/master/nidaqmx/tests/test_stream_digital_readers_writers.py

Please let us know if you still have any issues.

neilvana, 2018-03-06:
Is this still an issue?  Can I close this?

MaxP92, 2018-03-10:
no, 'write_one_sample' has not the optional 'auto_start' argument