ni/nidaqmx-python #48: Documentation should encourage restricing install to compatible versions

epage, 2018-01-26:
This project is following semver.  The documentation should encourage the user to follow best practices, including helping them from accidentally upgrade to an incompatible version until they are ready

For example, instead of 
```bash
$ python -m pip install nidaqmx
```
have
```bash
$ python -m pip install nidaqmx~=0.5.0
```

According to [PEP 440](https://www.python.org/dev/peps/pep-0440/#compatible-release), the compatible release version specifier will translate from `~=0.5.0` to `>=0.5.0,==0.5.*`

epage, 2018-01-26:
The downside is that you'll need to remember to update this.

nixnet-python has a [MAINTAINING](https://github.com/ni/nixnet-python/blob/master/MAINTAINING.rst) to document the release process.  Having something similar could help with this.

zhindes, 2024-01-18:
We don't need to teach people semver. Closing.

closed 2024-01-18 by zhindes