ni/nidaqmx-python #49: READ_ALL_AVAILABLE don't work; in_stream.avail_samp_per_chan always return 0

Sup3rGeo, 2018-02-21:
I spent a good couple of days trying to make the code run with AcquisitionType.CONTINUOUS and READ_ALL_AVAILABLE but nothing seemed to work.

After a lot of time wondering if it was something related to the acquisition, I could finally pinpoint the problem to the read function.

This code works fine:
```
data = None
task = nidaqmx.Task()
task.ai_channels.add_ai_voltage_chan("Dev1/ai1")
task.timing.cfg_samp_clk_timing(rate=60, sample_mode=AcquisitionType.CONTINUOUS)
task.start()
time.sleep(5)
t0=time.ctime()
data = task.read(300)
print "{}\n{}".format(time.ctime(), t0)
task.stop()
task.close()
print data
```
I know the buffer is already filled (and therefore not a problem with the acquisition) because the read function returns instantaneously, without waiting for data, and the data is what I expect.

This, however, don't:

```
data = None
task = nidaqmx.Task()
task.ai_channels.add_ai_voltage_chan("Dev1/ai1")
task.timing.cfg_samp_clk_timing(rate=60, sample_mode=AcquisitionType.CONTINUOUS)
task.start()
time.sleep(5)
t0=time.ctime()
data = task.read(READ_ALL_AVAILABLE)  # ONLY CHANGED FROM 300 to READ_ALL_AVAILABLE
print "{}\n{}".format(time.ctime(), t0)
task.stop()
task.close()
print data
```
And I get an empty list for my `data` variable.

So I read the source code and apparently the avail_samp_per_chan property (function) always return zero, for instance right before the read(300) command:
```
time.sleep(5)
print task.in_stream.avail_samp_per_chan
data = task.read(300)
```

So I believe that it is the problem. This function is not returning the correct number of available samples.

Can you confirm the problem and if so, is there any short-term workaround for this? This is a bit urgent as this is being used in a project that has already blew its deadline.

neilvana, 2018-03-06:
This is expected behavior due to how USB DAQ devices operate.  To achieve maximum throughput, samples are transferred from the USB device to the host device (your computer) in chunks.  Calling available_samp_per_chan only indicates the number of samples already available on your host machine.  In other words it is the number of samples that are immediately available if you call read.

When you specify a specific number of samples when calling a read the driver to send a message to the USB device telling it to immediately flush the requested samples to the host making them available.  The function blocks until the samples are transferred.  When you ask for only available samples, the read will return immediately with 0 samples initially but if continued will eventually return samples.  The driver sends a message to the USB device to start flushing its buffer when you execute the first read available samples if no samples are currently available.

neilvana, 2018-03-06:
This is not a issue with the driver or the Python api.

closed 2018-03-06 by neilvana

Sup3rGeo, 2018-03-07:
@neilvana thanks for the explanation!