ni/nidaqmx-python #50: TDMS recording

jayache80, 2018-03-05:
The examples don't show how to log long recording data to a tdms file. I'm assuming this API can do that. Anyone have a hello world for that? 

Currently I've set it up to register a callback for every N samples to append some list, as in the every_n_samples_event.py examples. At the end of the script, I write that list to a file. The problem comes when I'm now trying parse all the different channels, they're separated in chunks of N and I'd like to just use the work NI already did with TDMS.

neilvana, 2018-03-06:
The DAQmx API allows configuring the task to dump right to a file without having to need.  For example you can do the following:

`import nidaqmx
import nidaqmx.constants
import time

if __name__ == "__main__":
    with nidaqmx.Task() as task:
        task.ai_channels.add_ai_voltage_chan(
            "PXI1Slot2/ai0")
        task.in_stream.configure_logging(
            r"c:\Users\nstoddar\Desktop\task.tdms",
            nidaqmx.constants.LoggingMode.LOG)
        task.timing.cfg_samp_clk_timing(
            1000,
            samps_per_chan=1000)
        task.start()
        while not task.is_task_done():
            time.sleep(0.5)
`
This example is just a single finite acquisition for 1 second but will log all the data to disk.  You can alternatively choose LOG_AND_READ to cause the driver to log to the file every time you call read.  This is lower performance but allows you to both log the data and access it directly in the program.

chuyuanver, 2018-03-09:
Thanks for posting this example. I've tried this code but received an error message: The specified resource is reserved. The operation could not be completed as specified.
Do you have any suggestions about this?

jayache80, 2018-03-10:
Thanks so much for this example. I'm not sure how I missed the `configure_logging` when I was browsing the API. I got it to work wonderfully. Here is my code:
```
import nidaqmx
from nidaqmx.constants import (
    LineGrouping,
    TerminalConfiguration,
    VoltageUnits,
    AcquisitionType
    )
import time

# length of time (in seconds) to record data
record_time = 5

system = nidaqmx.system.System.local()
print(system.driver_version)
for device in system.devices:
    print(device)

with nidaqmx.Task("reader_task") as reader:
    reader.ai_channels.add_ai_voltage_chan(
        "Dev1/ai0:7",
        terminal_config=TerminalConfiguration.RSE,
        min_val=-5.0,
        max_val=5.0,
        units=VoltageUnits.VOLTS,
        )

    reader.in_stream.configure_logging(
        r"c:\daq_logs\reader_task.tdms",
        nidaqmx.constants.LoggingMode.LOG)

    # samps_per_chan is buffer size in CONTINUOUS mode
    # if in FINITE mode, samps_per_chan is number of samples to acquire
    reader.timing.cfg_samp_clk_timing(
        rate=1000,
        sample_mode=AcquisitionType.CONTINUOUS,
        samps_per_chan=2048,
        )

    input("Press <enter> to start recording")

    reader.start()

    time.sleep(record_time)

    reader.stop()
```

zhindes, 2024-01-18:
Closing this due to inactivity. If anybody wants to contribute, feel free :)

closed 2024-01-18 by zhindes