ni/nidaqmx-python #52: nidaqmx memory leak

dolhop, 2018-03-22:
version: nidaqmx 0.5.7
NI libraries: nidaq core 9.6

I've been using v9.6 for years.  Recently changed from PyNidaqMx to nidaqmx and noticed a memory leak when polling a digital input line.  The below code, while extreme, exacerbates the issue.

`'''test to show memory leak in nidaqmx'''
# using a USB-6009
# NidaqCore 9.6 library

import nidaqmx

daq = nidaqmx.system.System().devices[0]

while True:
    with nidaqmx.Task('task') as task:
        task.di_channels.add_di_chan(lines=daq.name + '/port0/line3')
        task.start()
        result = task.read()
`

secretaliasname, 2019-01-22:
I am also having this issue. I think part or all of the problem may be the __del method of Task. Apparently implementing the __del__ method on an object that contains a circular reference prevents python from garbage collecting properly. I am able to make the memory leak go away by commenting out the __del method of Task. It would be great to have an official fix for this. I am using python 2.7 but this may behave differently in >3.4 which has purportedly fixed this language wide in PEP-442. 

```
def __del__(self):
    if self._handle is not None:
        warnings.warn(
            'Task of name "{0}" was not explicitly closed before it was '
            'destructed. Resources on the task device may still be '
            'reserved.'.format(self.name), DaqResourceWarning)
```

secretaliasname, 2019-01-22:
There is probably a special place in hell for doing this, but I was able to work around the issue by:
```
import nidaqmx
del nidaqmx.Task.__del__
```

KennethNielsen, 2019-04-01:
@neilvana any update on this issue?

zhindes, 2022-02-10:
I can no longer reproduce this. I believe it was fixed in https://github.com/ni/nidaqmx-python/commit/76134740e919f476884e40d3ef6f4a4574861078. Please reopen with a reproducing failure case with the most recent release.