ni/nidaqmx-python #61: Control digital output timing

carlodri, 2018-05-30:
To generate a single digital pulse on a NI PCIe-6536 I am using the following code:

```python
ack_task = nidaqmx.Task()
ack_task.do_channels.add_do_chan("/Dev1/port4/line2")
while <some condition>:
   ack_task.write([True, False], auto_start=True)
   <read stuff from another channel>
ack_task.close()
```

The problem I experience is that sometimes the pulse width varies in a random way, _i.e._ it is mostly 1us long (which is what I need), but sometimes randomly jumps above 5us. What am I doing wrong? Is there a way to explicitly set the timing of the pulse? I tried using `cfg_samp_clk_timing` but then it complains that:

```
nidaqmx.errors.DaqError: Task contains a physical channel not supported by this device, given the requested Sample Timing Type.

To keep the Sample Timing Type, use physical lines from port0/line0 through port3/line7.  To access the requested channel, change the Sample Timing Type.
Property: DAQmx_SampTimingType
Requested Value: DAQmx_Val_SampClk

Output Channels: Dev1/port4/line2
```

Please note that using LabVIEW, using a digital waveform, this problem does not occur (_i.e._ the pulse duration is always 1us)

neilvana, 2018-08-01:
What device are you using?  This code is using software timing, so depending on other things going on your system there can be significant jitter.  To eliminate jitter you will need to use hardware timing which uses an onboard clock on the board to time the update of the digital channels.  This however, requires features which only exist on some products.

neilvana, 2018-08-02:
The PCIe-6536 does support hardware timing but only on ports 0-3.  Port 4 is PFI lines which can be used for either software timed static IO (which is what you are doing above) or for importing or exporting timing signals (i.e. sample clock, sample clock timebase).

See http://zone.ni.com/reference/en-XX/help/371893D-01/6536and6537help/frontpanel_6535_6/

If you need deterministic timing you should switch to a pin on ports 0-3.  You will not be able to achieve highly deterministic timing on Port 4 since system jitter will dominate.  You can make it more deterministic by raising the priority of the thread doing the software time IO, but it will still have pretty significant jitter and may not meet the requirements for your application.

carlodri, 2018-08-08:
no problem, thanks @neilvana  for the heads up! I'll look into this, and in the meanwhile I'm closing this.

closed 2018-08-08 by carlodri