ni/nidaqmx-python #62: Question about streaming a sinewave signal

WilyMongooseZero, 2018-06-29:
I'm trying to generate a sine wave with a stream writer, and cannot get the signal to generate more than once even though I have continuous flagged in the sample clock.
with other ao channels I can write to the channel and the static value stays after exiting the with statement, but when using stream writers i cannot get the signal to sustain itself as I can with labview. The only way I have managed it is to create a while loop which locks up my python program while executing.

I create the numpy array with one cycle of the sinewave, and then want to trigger this continuously, but the write_many_sample code blocks do not seem to do this.

````
SamplesPerSignal = 1000
SignalFrequency=60
#arbitrarily large number of cycles attempting to generate the signal many times does not work here.
NumberOfCycles = 100000000
Amplitude=8.9
w=2*np.pi*SamplesPerSignal
Sample = np.linspace(0,SamplesPerSignal,num=SamplesPerSignal)
Signal= np.sin(Sample*w)*Amplitude
with nidaqmx.Task() as task:
 task.ao_channels.add_ao_voltage_chan(MyDevice)
 task.timing.cfg_samp_clk_timing(rate=SignalFrequency*SamplesPerSignal,sample_mode=nidaqmx.constants.AcquisitionType.CONTINUOUS,samps_per_chan=SamplesPerSignal*NumberOfCycles)
 SignalStreamer = nidaqmx.stream_writers.AnalogSingleChannelWriter(task.out_stream, auto_start=True)
 SignalStreamer.write_many_sample(Signal)
```
This doesn't work at all, and the NumberOfCycles value seems to be ignored, as throwing a print statement around the write_many_samples just returns the number of samples in Signal.

WilyMongooseZero, 2018-06-29:
I can throw a while loop to a limit to generate the Signal continuously, but this is counter productive, as I need to trigger the signal and do other stuff not loop feed a sinewave.
```
Limit = 0
while Limit<=NumberOfCycles:
 SignalStreamer = nidaqmx.stream_writers.AnalogSingleChannelWriter(task.out_stream, auto_start=True)
 SignalStreamer.write_many_sample(Signal)
 Limit+=1
```
Generates the signal, but keeps the program busy generating the signal.

czimm79, 2019-08-08:
I know this question is over a year old, but in an effort to help people like me searching for the answer to this question:

Using `with nidaqmx.Task() as task` will try to end the task once it "thinks" its been completed. Solve this by simply setting `task = nidaqmx.Task()` and then adding logic at the end of the code to stop and clear the task yourself.

zhindes, 2022-02-08:
This is a general NI-DAQmx problem and not an issue with the `nidaqmx` Python bindings. Consider posting on https://forums.ni.com/ for help resolving issues such as this.

closed 2022-02-08 by zhindes