ni/nidaqmx-python #66: Include more information on stream writers and readers

bburan, 2018-08-16:
It's unclear to me what type of performance benefits we get using the stream writers and readers are. Can you please elaborate (in the documentation) what type of tricks are used to get a performance boost and what type of boost we might expect to see?

bburan, 2022-02-08:
@zhindes Was this addressed in the documentation? I reviewed the commit log and could not find any updates to the documentation that address this issue. It'd be great to learn more about why one should consider using stream writers and readers.

zhindes, 2022-02-08:
The [documentation](https://nidaqmx-python.readthedocs.io/en/latest/) says:

> Consider using the nidaqmx.stream_readers and nidaqmx.stream_writers classes to increase the performance of your application, which accept pre-allocated NumPy arrays.

That's the crux of it. You define your arrays and we will use them versus creating a new array on the fly like we do [here](https://github.com/ni/nidaqmx-python/blob/f58645f66503844bea9c6a8eb905d7b1eef7bd7c/nidaqmx/task.py#L620).

bburan, 2022-02-08:
@zhindes Thanks for the clarification.