ni/nidaqmx-python #68: Filter properties not supported in nidaqmx-python

hertrich, 2018-09-27:
The following NI-DAQmx properties are not supported in nidaqmx-python:
Analog Input >> General Properties >> Filter >> Enable
Analog Input >> General Properties >> Filter >> Frequency
Analog Input >> General Properties >> Filter >> Response
Analog Input >> General Properties >> Filter >> Order
Those properties should be introduced in _task_modules/channels/ai_channel, next to 
Analog Input >> General Properties >> Filter >> Filter Delay
and 
Analog Input >> General Properties >> Filter >> Filter Delay Units.

rvnfay, 2020-01-31:
I have the same issue. Not sure how to go about it atm, but I think I should find a way to use C for this part of task.

zhindes, 2022-02-09:
Fixed in #136