ni/nidaqmx-python #69: Documentation search results return "SORRY This page does not yet exist"

dolorsitatem, 2018-10-03:
This appears to be the case for all entries.

For example,

[](https://nidaqmx-python.readthedocs.io/en/latest/index.html)
Search for: nidaqmx.constants.ACExcitWireMode
Results are found, the first one being for nidaqmx.constants.ACExcitWireMode
Clicking on that link sends you to the error page

wireddown, 2018-11-26:
[image]

KennethNielsen, 2019-04-01:
The issue seems to be that there is a ".rst" in injected in the URI that shouldn't be there, i.e. the links goes to something like `.....task.rst.html....` where they should go to `.....task.html....`

KennethNielsen, 2019-04-01:
AS outlines e.g. in this issue (https://github.com/rtfd/readthedocs.org/issues/2708), this seems to be related to a bug in sphinx 1.5 and the work-around for now is to downgrade the version.

KennethNielsen, 2019-05-05:
I just wanted to bump this. The docs are not easy to use with this issue and if the solutions for now is to lock the sphinx versions at a lower version it should be a relatively simple fix.

zhindes, 2022-02-09:
This appears to be resolved with our latest update to dependencies.

closed 2022-02-09 by zhindes