ni/nidaqmx-python #73: DeprecationWarning in Python 3.7

robbmcleod, 2018-12-21:
In 3.7 a deprecation warning which will become an error in 3.8 appears:

    from collections import Sequence
    c:\python37\lib\site-packages\nidaqmx\system\_collections\physical_channel_collection.py:8: DeprecationWarning: Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated, and in 3.8 it will stop working

gadeberg, 2020-01-15:
The 'will stop working' is currently postponed to 3.9 (hope NI will update soon):

```
\.venv\lib\site-packages\nidaqmx\system\_collections\physical_channel_collection.py:8: DeprecationWarning: Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated since Python 3.3, and in 3.9 it will stop working
    from collections import Sequence
```

tirkarthi, 2020-08-29:
The warning has been postponed again to 3.10 . I have created #104 for this issue.

danousley, 2020-08-31:
Thanks for the PR @tirkarthi. The team will look over it.