ni/nidaqmx-python #74: generate defragged tdms files

baudoux, 2019-01-31:
The tdms files generated by the stream are heavy to open, probably because they are not defragged.
I generate the file with:

`task.in_stream.configure_logging(file_path, nidaqmx.constants.LoggingMode.LOG_AND_READ, 'myGroupName', nidaqmx.constants.LoggingOperation.OPEN_OR_CREATE)
`
Is it possible to generate defragged files?

AndrewHeim, 2020-11-25:
I don't think this is out of the norm for any sort of TDMS log.  Defragging is typically done post-facto.  Alas - I don't know of a python call to do so.  Maybe you could call some LabVIEW code (built as an exe) from python to do it?

zhindes, 2022-02-08:
This is a feature request of the DAQmx API and not of the `nidaqmx` Python bindings. Consider making a request on https://forums.ni.com/.

closed 2022-02-08 by zhindes