ni/nidaqmx-python #76: ModuleNotFoundError: No module named 'nidaqmx'

aherting, 2019-03-29:
I've recently downloaded Python 3.73. I'm hoping to use the nidaqmx library to collect voltage data. I was trying out the example in the nidaqmx page, but get the following error. Does anyone have any ideas why? Thanks in advance for your help. 

ModuleNotFoundError: No module named 'nidaqmx'

Below is the example I tried to run:
import pprint
import nidaqmx

pp = pprint.PrettyPrinter(indent=4)

with nidaqmx.Task() as task:
    task.ai_channels.add_ai_voltage_chan("cDAQ1Mod1/ai0")

    print('1 Channel 1 Sample Read: ')
    data = task.read()
    pp.pprint(data)

    data = task.read(number_of_samples_per_channel=1)
    pp.pprint(data)

    print('1 Channel N Samples Read: ')
    data = task.read(number_of_samples_per_channel=8)
    pp.pprint(data)

    task.ai_channels.add_ai_voltage_chan("cDAQ1Mod1/ai1:3")

    print('N Channel 1 Sample Read: ')
    data = task.read()
    pp.pprint(data)

    print('N Channel N Samples Read: ')
    data = task.read(number_of_samples_per_channel=2)
    print (data)

MarcusTorndahl73, 2019-04-11:
Have you installed (pip | easy_install) the 'nidaqmx' package?

aherting, 2019-04-11:
I figured it out! The problem was that the security on my computer was giving me troubles when trying to download the library ‘nidaqmx.’ I do have a different question though about getting an application using the ‘nidaqmx’ import to continuously scan/update the program. Are you familiar with how to do that? The first one that establishes the task and the second one that evaluates the result. Basically, when I first compile the program, the result in the moment of compiling appears, but when I change the voltage going to the pin, the application does not update. Below are just two snippets of the code I have. Do you have any thoughts on this? Thanks!

#Establish the port that I want to read data from

with nidaqmx.Task() as task:
    task.ai_channels.add_ai_voltage_chan("Dev1/ai0")
    data = task.read()

#If the correct voltage is seen then we want a green circle to appear else a red one will appear.

  while True:
        #Minimum voltage expected to be seen
        min_val = 10
        if data > min_val:
            grnCircle()
        elif data < min_val:
            redCircle()

thoagland, 2019-07-16:
I'd suggest that this issue be closed

rares-pop, 2020-05-09:
Yes, seems this isn't really an issue, but a config problem.

closed 2020-05-09 by rares-pop