ni/nidaqmx-python #77: Install into Docker image

lazzarello, 2019-04-18:
I'm trying to install this library into a Windows server 2019 Docker container. I have the base OS and Python 3.7.3 installed. The library loads into an interactive REPL but when any functions are called that get passed to nicaiu.dll I get the following exception

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "C:\Python\lib\site-packages\nidaqmx\system\system.py", line 73, in driver_version
    return DriverVersion(self._major_version, self._minor_version,
  File "C:\Python\lib\site-packages\nidaqmx\system\system.py", line 108, in _major_version
    cfunc = lib_importer.windll.DAQmxGetSysNIDAQMajorVersion
  File "C:\Python\lib\site-packages\nidaqmx\_lib.py", line 166, in windll
    self._import_lib()
  File "C:\Python\lib\site-packages\nidaqmx\_lib.py", line 209, in _import_lib
    windll = ctypes.windll.LoadLibrary(lib_name)
  File "C:\Python\lib\ctypes\__init__.py", line 434, in LoadLibrary
    return self._dlltype(name)
  File "C:\Python\lib\ctypes\__init__.py", line 356, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: [WinError 126] The specified module could not be found
```
I copied some DLL's from the host system manually but this was not enough for ctypes to find the file. I'm not sure how to provide this file to the container so the code in `_lib.py` finds the DLL.

lazzarello, 2019-04-23:
I've tried seven different installation methods and all have failed. [I wrote up a post on the NI forum](https://forums.ni.com/t5/Multifunction-DAQ/Python-ctypes-cannot-import-nicaiu-dll/td-p/3918544) with my findings.

At the moment it seems impossible to run code that calls functions from nidaqmx-python in a Docker image.

AndrewHeim, 2020-11-25:
In my understanding, it may be possible to install this package... but it won't be possible to run it.  It's not the fault of NI or this repo, either - it is simply standard practice that the only access a docker container has to the outside world is to volumes and the network.  I don't know of any other docker or container solution anywhere that has access to hardware drivers.

I'm very interested in seeing LabVIEW in a docker image, but mostly for CI purposes.  I don't expect driver calls to be more than stubbed in such an environment.

zhindes, 2022-02-08:
The `nidaqmx` Python bindings require the NI-DAQmx driver installed.

closed 2022-02-08 by zhindes

jpwright, 2023-05-03:
To share my experience- I was able to `nidaqmx` working (at least to the point of being able to import the package and create a task), with these conditions:

- Ubuntu 22.04 as both host OS and container base image, `ni-daqmx` driver installed in both
- Container run in `--privileged` mode
- Container run with host networking (`--network host`)

rbuckland, 2023-05-15:
The nidaqmx drivers seem to require `systemctl` installed to operate. 
```
RUN   dpkg -i NILinux2023Q2DeviceDrivers/ni-ubuntu2204-drivers-2023Q2.deb && \
    apt update && \
    DEBIAN_FRONTEND=noninteractive \
    apt-get -y install libnidaqmx libnidaqmx-devel
```

and it errors with 

```
#6 19.14 /var/lib/dpkg/info/ni-pxiplatformservices-bin.postinst: 6: systemctl: not found
#6 19.14 dpkg: error processing package ni-pxiplatformservices-bin (--configure):
```

There are a number of packages that, on post install, need to run systemctl.
I have a few thoughts (re-write the packages, fake the systemctl, leverage - https://github.com/gdraheim/docker-systemctl-images) 

@jpwright  - how did you work around this ?

rbuckland, 2023-05-15:
Actually, the python systemctl from gdraheim, did work for the install. 

```
COPY .devcontainer/systemctl3.py /usr/bin/systemctl
RUN test -L /bin/systemctl || ln -sf /usr/bin/systemctl /bin/systemctl

RUN ...  # get driver 
    && \
    unzip NILinux2023Q2DeviceDrivers.zip && \
    dpkg -i NILinux2023Q2DeviceDrivers/ni-ubuntu2204-drivers-2023Q2.deb && \
    apt update && \
    DEBIAN_FRONTEND=noninteractive \
    apt-get install -y --no-install-recommends libnidaqmx libnidaqmx-devel
```