ni/nidaqmx-python #84: Is this repository still being maintained?

czimm79, 2019-08-05:
There are a lot of open bugs that haven't seen any updates in over a year. Has National Instruments discarded this project?

zhindes, 2019-08-23:
Hi czimm79,

I apologize for the lack of activity on this project in the past year or so. The responsibility for this project is in the process of being transferred to a new team within National Instruments, and we expect that team to begin actively maintaining this API soon. In the interim, please continue to post any issues that you may have found. We look forward to continuing to improve our Python support for nidaqmx.

Thanks!
Zach

simonfagerholm, 2020-01-23:
Hi @zhindes!
So, half a year later, how is the new team progressing? Any time plan that can be shared with us?

Best regard, Simon

zhindes, 2020-01-23:
It is still in flight. We hope to be able to share more soon.

nzjrs, 2020-04-06:
Sorry to bother, but I guess things are still moving here?

danousley, 2020-04-06:
Hi @nzjrs, thanks for pinging. We're actively standing up the backlog for the team that just transitioned into ownership for this repo. I will provide an update on or before April 15.

danousley, 2020-04-16:
1) In the short term, we still have at least a couple more priority decisions to get through before we can say where maintenance on this repo will be in our backlog. This has been an active topic for us in the past week and I'm expecting we will have agreement somewhere in the range of a few days to couple of weeks.
2) In the longer term, we are building on the work already completed for this v0.5 DAQmx repo with a goal to release a 1.0 version.

rares-pop, 2020-05-11:
@nzjrs - do you have something in particular that needs to be solved?

syl49, 2021-03-15:
Have you some news for a 1.0 version ?? A date ?? Thanks

rares-pop, 2021-03-17:
@syl49 - no news in term of a deadline or ETA, but I can share some of the things we consider are needed to be done before calling it 1.0:
 - adding support for the timing operations (wasn't done before because of the lack of high precision timing libraries)
 - better documentation
 
 One thing on our plate but not necessarily for 1.0, is to have the low-level python api auto generated so it stays in sync with the actual drivers (C code).

Unfortunately, priorities shifted again and the major focus for NI right now is to support *all* drivers on LinuxRT, and I believe in the next few months, we will only put the fires out regarding this repo.

That being said, is something particular functionality you are looking for?

syl49, 2021-03-18:
Thanks you for response

Our team have post two problems found in developping our interface to drive a NI6501 , see "Nidaqmx Problem Read and Depreciation" , see topic issue add 3 days ago.
The problem is that nidaqmx actually is not compatible with python 3.9 due to  numpy version 1.19.5. To solve the problem, We have been created our own class inherit from nidaqmx and change method impacted to correct the depreciation..it's not very the best.
The other problem is to change the output discrete..something this very strange in function write , see my topic issue, if you need more information, ask me.

Syl

PhilReinhold, 2021-08-14:
Are you planning on making a new release at some point? I was looking into an issue, and saw that the fix was pushed almost a year ago https://github.com/ni/nidaqmx-python/pull/104, but there have been no new versions released in 3 years.

danousley, 2021-08-18:
@PhilReinhold Hi, sorry to leave you hanging. Thanks for bringing this up. The release-work crossed some team boundaries and got starved and then slipped through the cracks. But it is overdue for being addressed with a release. We are looking to re-align internally so that we can sustainably get fixes pushed and publish new versions for sustaining the support we already offer. 

Additional development beyond sustaining is still on hold for ni-daqmx-python at the moment while we fill out our Linux OS offerings further.

zhindes, 2022-02-09:
This repo is now better maintained and has had multiple releases recently with a few more expected.

closed 2022-02-09 by zhindes