ni/nidaqmx-python #85: How to locate "nicaiu. dll" in my computer ?

WuTUT, 2019-09-11:
My computer version is windows10 (64 bits), and I install labview2018 64 bits, NI MAX 18.0.
I want to know the file directory of nicaiu.dll which nidaqmx-python uses.
So how can I get it?

`print(sys.version)` returns:
`3.6.9 |Anaconda, Inc.| (default, Jul 30 2019, 14:00:49) [MSC v.1915 64 bit (AMD64)]`

rgerkin, 2019-12-17:
On my machine it was at `C:/Windows/System32/nicaiu.dll`

rares-pop, 2020-05-05:
The code that's loading the file is here:
https://github.com/ni/nidaqmx-python/blob/master/nidaqmx/_lib.py#L209

So, yes, it's looking in the standard search path. That includes the PATH environment variable if it is set (you can check in CMD by typing 'set' and check for PATH).

On my win 10 64bit, it's the same location @rgerkin mentions - 

> C:/Windows/System32/nicaiu.dll

closed 2020-05-05 by rares-pop