ni/nidaqmx-python #91: `Task.__del__` broken

spalatofhi, 2020-04-10:
The `__del__` method doesn't work. It tries to provide a useful warning message, but instead more error messages are generated. This can give rise to unnecessary errors, or to obfuscate the actual source of error.

Examples:

Doing nothing: 
```python
from nidaqmx import Task
t = Task()
```
yields:
```
Exception ignored in: <function Task.__del__ at 0x0000029D04F3DD30>
Traceback (most recent call last):
  File "C:\Users\samue\Anaconda3\envs\trion_dev\lib\site-packages\nidaqmx\task.py", line 105, in __del__
  File "C:\Users\samue\Anaconda3\envs\trion_dev\lib\site-packages\nidaqmx\task.py", line 156, in name
  File "C:\Users\samue\Anaconda3\envs\trion_dev\lib\site-packages\nidaqmx\errors.py", line 115, in check_for_error
ImportError: sys.meta_path is None, Python is likely shutting down
```
In this case, I would not expect an error, as I've done nothing. The `nidaqmx` library tries to warn us about failing to call `t.close()`, but trips in the process.

Correct way of doing nothing:
```python
from nidaqmx import Task
t = Task()
t.close()
```

Try doing something, but get an error.
```python
from nidaqmx import Task
t = Task()
raise RuntimeError("Didn't try hard enough")
t.close()
```
yields:
```
Traceback (most recent call last):
  File ".\task_error.py", line 4, in <module>
    raise RuntimeError("Didn't try hard enough")
RuntimeError: Didn't try hard enough
Exception ignored in: <function Task.__del__ at 0x000002053E3DDD30>
Traceback (most recent call last):
  File "C:\Users\samue\Anaconda3\envs\trion_dev\lib\site-packages\nidaqmx\task.py", line 105, in __del__
  File "C:\Users\samue\Anaconda3\envs\trion_dev\lib\site-packages\nidaqmx\task.py", line 156, in name
  File "C:\Users\samue\Anaconda3\envs\trion_dev\lib\site-packages\nidaqmx\errors.py", line 115, in check_for_error
ImportError: sys.meta_path is None, Python is likely shutting down
```
This traceback is unnecessarily complicated, making the original error harder to find.

For more complicated scenarios, the exact behavior depends on the order of garbage collecting.

Machine-Hum, 2020-04-20:
Any resolution on this? I'm having the exact same issue. I want to drop some outputs in my  dtor but this error is blocking me.

spalatofhi, 2020-04-20:
An official resolution would be to fix the bug in `Task.__del__`.

This error doesn't arise if you call `task.close()`. This can be ensured in a few ways.
- Wrap everything in a `with` or `try ... finally` block, as is done in the examples. If your code is somewhat complex, that may not be quick...
- Use a global `ExitStack`... (https://docs.python.org/3/library/contextlib.html#contextlib.ExitStack). This should fit better into existing code, but you now have to carry around an ExitStack.
- Monkeypatch the task class. This is not recommended in actual code, but just for fun:
Replace the `name` property:
```python
from nidaqmx import Task
t = Task()
Task.name = property(lambda self: self._saved_name)
raise RuntimeError("Didn't try hard enough")
t.close()
```
If you're feeling reckless, you can replace the whole `__del__` method, of all Task objects, as it doesn't do anything apart from encountering an error when trying to raise a warning...
```python
from nidaqmx import Task
t = Task()
type(t).__del__ = lambda *a: None # Don't *actually* do this!
raise RuntimeError("Didn't try hard enough")
t.close()
```

rares-pop, 2020-05-05:
@spalatofhi , @Machine-Hum - what setup do you guys use? I see it's windows, but what python version and what version of the driver?

spalatofhi, 2020-05-05:
Windows 10 64 bits.
nidaqmx = 0.5.7
numpy = 1.18.1
six = 1.14.0

```
PS C:\work\src\trion\launcher> python -c "import sys; print(sys.version)"
3.8.2 | packaged by conda-forge | (default, Mar  5 2020, 17:29:01) [MSC v.1916 64 bit (AMD64)]
PS C:\work\src\trion\launcher> python -c "import nidaqmx; print(nidaqmx.system.Sys
tem.local().driver_version)"
DriverVersion(major_version=19, minor_version=6, update_version=0)
```

Your call to `del t` does not cause the execution of Task.__del__. The task needs to be garbage-collected for this to happen.

From the REPL, your example raises a warning. 
```
>>> from nidaqmx import Task
>>> t = Task()
>>> del t
>>> exit()
C:\Users\samue\Anaconda3\envs\trion_dev\lib\site-packages\nidaqmx\task.py:102: ResourceWarning
: Task of name "_unnamedTask<0>" was not explicitly closed before it was destructed. Resources
 on the task device may still be reserved.
  warnings.warn(
```

However:
```
PS C:\work\src\trion\scratch\ni> python
Python 3.8.2 | packaged by conda-forge | (default, Mar  5 2020, 17:29:01) [MSC v.1916 64 bit (
AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> from nidaqmx import Task
>>> t = Task()
>>> exit()
Exception ignored in: <function Task.__del__ at 0x000002242FCA0D30>
Traceback (most recent call last):
  File "C:\Users\samue\Anaconda3\envs\trion_dev\lib\site-packages\nidaqmx\task.py", line 105,
in __del__
  File "C:\Users\samue\Anaconda3\envs\trion_dev\lib\site-packages\nidaqmx\task.py", line 156,
in name
  File "C:\Users\samue\Anaconda3\envs\trion_dev\lib\site-packages\nidaqmx\errors.py", line 115
, in check_for_error
ImportError: sys.meta_path is None, Python is likely shutting down
```

I note that this is not the error I reported. I was interested in the case where an exception is raised during execution, and the call to `del` is not reached.

rares-pop, 2020-05-05:
Gotcha! I see it reproducing on both windows and LinuxRT! Thank you for the help! I'll look into fixing this.

closed 2020-05-12 by rares-pop