ni/nidaqmx-python #107: Python  programmatic control of NI-DAQ Measurement & Automation Explorer

git-tm2020, 2020-09-24:
Upon discovering a faulty DAQ-9171 device and replacing it with another one, The NI-DAQ Measurement & Automation Explorer  Tray/App will not remove the previous "cDAQ1" DAQ-9171 device. If a replacement DAQ-9171 is inserted into the PC, it shows up as "cDAQ2" or "cDAQ2Mod2". My python code opens, "cDAQ1".  I can resolve this by deleting the original "cDAQ1" that is linked to the defective 9171 entry manually, but i would like to do this using Python. Is there an Python API to control, remove, and rename, device entries from The NI-DAQ Measurement & Automation Explorer and Windows OS programmatically by using Python ?

nturley, 2020-09-25:
I think what you want is nisyscfg or maybe SysApi, I always get the two confused. I thought there might be official python bindings for nisyscfg but I can't find them but it looks like several people have made unofficial ones like this one for example.

https://github.com/tkrebes/nisyscfg-python

git-tm2020, 2020-09-25:
> I think what you want is nisyscfg or maybe SysApi, I always get the two confused. I thought there might be official python bindings for nisyscfg but I can't find them but it looks like several people have made unofficial ones like this one for example.
> 
> https://github.com/tkrebes/nisyscfg-python

Thanks for the help!

closed 2022-02-08 by zhindes