ni/nidaqmx-python #109: Counter not working - time out

ziofitt, 2020-10-28:
I am using the following code to work with a counter (ctr0) but I receive a timeout error. Any thought? (the DAQ I am using is a PXIe 6361)

```
import nidaqmx
from nidaqmx.types import CtrTime

with nidaqmx.Task() as task_pulse:
    task_pulse.co_channels.add_co_pulse_chan_time("PXI1Slot8/ctr0", name_to_assign_to_channel="Pippo", initial_delay=0,low_time=0.01,high_time=0.01)    
    sample = CtrTime(high_time=0.01, low_time=0.01)
    task_pulse.write(sample)
    task_pulse.close()
```

**ERROR MESSAGE:**

> DaqError: Specified operation did not complete, because the specified timeout expired.
> Task Name: _unnamedTask<3F>
> 
> Status Code: -200474

danousley, 2021-01-19:
@ziofitt Did you ever resolve this?

ziofitt, 2021-01-19:
Unfortunately not. I ended up using a simple digital output :(

zhindes, 2022-02-08:
This is a general NI-DAQmx problem and not an issue with the `nidaqmx` Python bindings. Consider posting on https://forums.ni.com/ for help resolving issues such as this.

closed 2022-02-08 by zhindes