ni/nidaqmx-python #113: Select Output drive type NI USB 6501

MathiasR19, 2021-02-02:
Hi all, 
I want to configure output per output of my IO card. for example I want output p1.0 to be active drive and p1.1 to be open collector. I can’t figure out how to do that in python.
I can made it with Labview.
I found this on internet :
https://nidaqmx-python.readthedocs.io/en/latest/do_channel.html#nidaqmx._task_modules.channels.do_channel.DOChannel
or 
https://nidaqmx-python.readthedocs.io/en/latest/constants.html

I tried a lot of possibilities but i can't do it.
This is my software and my tests in comment : 

#Active_Drive = 12573
#Open_collector = 12574
#import nidaqmx
#system = nidaqmx.system.System.local()
#for device in system.devices:
#    devices = str(device)
#    print (devices)
#    print()
#task = nidaqmx.Task()
## nidaqmx.constants.DigitalDriveType(Active_Drive)
#task.do_channels.add_do_chan("CarteIO/port0/line0")
## nidaqmx.DOChannel.do_output_drive_type("CarteIO/port0/line0",Active_Drive)
##nidaqmx.channel.do.output_drive_type("CarteIO/port0/line0",Active_Drive)
## nidaqmx.task.DOChannel(Actvie_Drain)
#nidaqmx._task_modules.channels.channel.DOChannel.do_output_drive_type(Active_Drive)
#task.start()
#value1 = False
#print('Valeur de la sortie :', value1)
#task.write(value1)
#task.stop
#task.close()

If you know, how to do this, and il you can help me, that would be great. I think there are many people like me.

Thanks, 
Have a good day

syl49, 2021-03-15:
C'est pas ça que tu cherches
with nidaqmx.Task() as task:
	task.do_channels.add_do_chan(dev_str, line_grouping=LineGrouping.CHAN_FOR_ALL_LINES)

	# print(f"Drive type avant : {task.do_channels.do_output_drive_type}");
	for chnl in task.do_channels:
		# chnl.do_output_drive_type = nidaqmx.constants.DigitalDriveType.ACTIVE_DRIVE
		chnl.do_output_drive_type = nidaqmx.constants.DigitalDriveType.OPEN_COLLECTOR

rares-pop, 2021-03-18:
@MathiasR19 - is @syl49 helpful? Do you need more info?

MathiasR19, 2021-03-18:
Merci pour ta réponse, depuis le 2 février j'avais trouvé mais ta réponse me convient bien. Cela aidera forcément quelqu'un d'autres, c'est utile.
Merci.

closed 2021-03-18 by MathiasR19