ni/nidaqmx-python #115: Deprecation Warning according nidaqmx packet

kfirCom, 2021-02-24:
working with pycharm version: community 2019.2.3 64bit
working  with python 3.8
working with nidaqmx version 0.5.7

in the last day my program not run and end with displayed followed:

 c:\python38\lib\site-packages\nidaqmx\system\_collections\physical_channel_collection.py:8: DeprecationWarning: Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated since Python 3.3, and in 3.9 it will stop working
    from collections import Sequence

I would like inform the program run well until tomorrow!

Regards

tirkarthi, 2021-02-24:
This should be fixed with https://github.com/ni/nidaqmx-python/pull/104

rares-pop, 2021-03-18:
This should be fixed. But for that, you need to use the latest source (i.e. git clone, and not pip install).

We were waiting for more small fixes like this before creating a small incremental release.

closed 2021-03-18 by rares-pop