ni/nidaqmx-python #129: nidaqmx is broken on python 3.10

erik-korzon-blueorigin, 2021-11-17:
Calling "import nidaqmx" on python 3.10 throws an ImportError, when the module calls "from collections import Sequence". Rolling back to python 3.9.9 fixes the issue. Directly calling "from collections import sequence" on python 3.9.9 gives the following depreciation warning:
`DeprecationWarning: Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated since Python 3.3, and in 3.10 it will stop working`

This module needs an update to retain compatibility with the current version of python.

MichaelARiley, 2021-11-30:
This issue is actually already fixed in the repository, but the fixed version does not appear to have been released or pushed to the pip repository.

If you clone the current github master (as of 2021-11-30) and install that version, or if you upgrade directly from github, you should get a version of nidaqmx that will work with Python 3.10.

zhindes, 2022-02-08:
This has been released.

closed 2022-02-08 by zhindes