ni/nidaqmx-python #142: How to write multiple sample pulse frequency for user time duration

giorgiamarcelli, 2022-02-08:
Hi, I wrote the code below in order to be able to connect with my NI instrument - The code is able to generate one pulse and then iterate the pulse generation for time t if setting time.sleep(t).  
I would like to be able to write multiple pulses for a time which is user's defined (I need to create a GUI in which the user pushes a button and create pulses for a random time and when he/she unclicked it the pulses stop). I noticed I cannot do it using CounterWriter.write_one_sample_pulse_frequency neither with CounterWriter.write_multiple_sample_pulse_frequency (since it requires to enter an array of a given length). Notice that it's very important for me to specify the frequency..

``
def generate_pulse(event):
    
    running = True
    
    while running :
    
        with nidaqmx.Task() as task:

            task.co_channels.add_co_pulse_chan_time("Dev1/ctr0")
   
            task.timing.cfg_implicit_timing(sample_mode=AcquisitionType.CONTINUOUS)
            cw =CounterWriter(task.out_stream, True)
            task.start()
    
            cw.write_one_sample_pulse_frequency(1., 0.50)    #frequency, dutycicle, timeout
            #time.sleep(50)
            
    quit()`

zhindes, 2022-02-08:
The [X Series User Manual](https://www.ni.com/pdf/manuals/370784k.pdf) has a great dissertation on counters and how they work. This applies to cDAQ, too (except some doesn't for 9172). Perhaps one of the applications they describe there is what you're looking for?

> Is it indeed possible to write pulses of a given frequency for a user specified time -  without calling time.sleep() within the code 

You could set it to `AcquisitionType.FINITE` for a set number of pulses (so user specified time divided by pulse width). Then each time the user hits the button you change the pulse specification and number of samples and restart the task?

zhindes, 2022-02-08:
Closing this as not an issue with the `nidaqmx` python API.