ni/nidaqmx-python #143: Event callbacks are clunky

maxxboehme, 2022-02-08:
[Original discussion on NI's forms](https://forums.ni.com/t5/Multifunction-DAQ/Python-DAQmx-bindings-get-task-from-task-handle/m-p/4206371/highlight/false#M102009).

DAQmx's event callback signatures include a Task ID and callbackData parameters even though there currently isn't a way to get a task object from a Task ID or be able to pass in user provided callbackData ([register_every_n_samples_acquired_into_buffer_event passes None as the data](https://github.com/ni/nidaqmx-python/blob/f58645f66503844bea9c6a8eb905d7b1eef7bd7c/nidaqmx/task.py#L891).)

Having to specify these to parameters in the signature even though they are not usable is confusing the users.

These are 2 ideas on how to improve this.
-  Provide a way to pass in user provided data and get a task object from a Task ID.
- Make the callback signature that users provide not have to specify those and wrap their callback in an internal one to the event register functions.

joshSonera, 2022-08-31:
Is there really no way to pass user variables in for callback_data? That would be unfortunate.

maxxboehme, 2022-08-31:
You can reference variables in the callback by creating an inner function (also called nested function) and Python allows you to reference variables from the enclosing function. You can see this being done in the [every_n_samples_event.py](https://github.com/ni/nidaqmx-python/blob/master/nidaqmx_examples/every_n_samples_event.py) example with the "callback" inner function referencing "samples" from the enclosed function.

joshSonera, 2022-08-31:
Ahh I see, thank you very much!

JoKeyser, 2024-07-25:
Thanks, this helped me as well.

For anyone else reading this in the future, here the perma-link to the code version referenced above by @maxxboehme :

https://github.com/ni/nidaqmx-python/blob/caf1b1cede57b5413d96077ba3ee61e243d83547/nidaqmx_examples/every_n_samples_event.py#L21

bkeryan, 2024-07-25:
FYI, here are two ways to pass user variables to the callback without using local functions (untested):
- Use [functools.partial](https://docs.python.org/3/library/functools.html#functools.partial)
  ```
  def callback_with_extra_param(extra_param, task_handle, every_n_samples_event_type, number_of_samples, callback_data):
      ...

  callback = functools.partial(callback_with_extra_param, 123)
  task.register_every_n_samples_acquired_into_buffer_event(1000, callback)
  ```
- Write a class with a `__call__` method
  ```
  class CallbackWithExtraParam:
      def __init__(self, extra_param):
          self.extra_param = extra_param

      def __call__(self, task_handle, every_n_samples_event_type, number_of_samples, callback_data):
         ...
 
  callback = CallbackWithExtraParam(123)
  task.register_every_n_samples_acquired_into_buffer_event(1000, callback)
  ```

(Still, we should fix this.)