ni/nidaqmx-python #162: Example for synchronizing tasks / read and write operations

fhchl, 2022-03-09:
It would be great to have some examples for how synchronize tasks. 

For example, I am have connected the output 0 of my USB-4431 to the input 0 and run

```python
import numpy as np
import nidaqmx as ni
from nidaqmx.constants import TaskMode

def query_devices():
    local_system = ni.system.System.local()
    driver_version = local_system.driver_version

    print('DAQmx {0}.{1}.{2}'.format(driver_version.major_version, driver_version.minor_version,
                                    driver_version.update_version))

    for device in local_system.devices:
        print('Device Name: {0}, Product Category: {1}, Product Type: {2}'.format(
            device.name, device.product_category, device.product_type))

def playrec(
    outdata, sr=96000, input_mapping=['Dev2/ai0'],
    output_mapping=['Dev2/ao0']
):
    nsamples = outdata.shape[0]
    with ni.Task() as read_task, ni.Task() as write_task:
        for o in output_mapping:
            aochan = write_task.ao_channels.add_ao_voltage_chan(o)
            aochan.ao_max = 3.5   # output range of USB-4431
            aochan.ao_min = -3.5
        for i in input_mapping:
            # FIXME: doesn't support multiple output mappings yet.
            aichan = read_task.ai_channels.add_ai_voltage_chan(i)
            aichan.ai_min = -10
            aichan.ai_max = 10

        for task in (read_task, write_task):
            task.timing.cfg_samp_clk_timing(rate=sr, source='OnboardClock', samps_per_chan=nsamples)
        
        # commiting task reduces lag between read and write
        read_task.control(TaskMode.TASK_COMMIT)
        write_task.control(TaskMode.TASK_COMMIT)

        write_task.write(outdata, auto_start=False)
        write_task.start()
        indata = np.asarray(read_task.read(nsamples)).T
        
    return indata

if __name__ == "__main__":
    query_devices()

    sr = 96000
    duration = 1
    t = np.linspace(0, duration, sr*duration, endpoint=False)
    sig = 1 * np.sin(1000*2*np.pi*t)

    indata = playrec(sig)
    import matplotlib.pyplot as plt
    plt.plot(t, indata)
    plt.show()
```

Looking at the plot, I can clearly see that there is around 5 ms of lag between the start of the write and the start of the read task on the device. 

If I run a similar example in Matlab, there is no such lag:

```matlab
duration = 10;
sr = 96000;         % NI 4431

t = (0:(duration*sr-1)) / sr;
sig = sin(t*1000*2*pi)';

% prep daq card
d = daq.getDevices;
s = daq.createSession('ni');
s.Rate = sr;

devname = 'Dev2';
outrange=[-3.5 3.5];
range_in=[-10 10];
 
outchan = s.addAnalogOutputChannel(devname,'ao0','Voltage');
outchan.Range = outrange;
voltchan = s.addAnalogInputChannel(devname,'ai0','Voltage');
voltchan.Range = range_in;

queueOutputData(s, sig);
data = startForeground(s);

indata = data(:, 1) 

plot(t,data(:,1));

s.release()
```

I already tried to replace the `write_task.start()` line with configuring the `write_task.triggers.start_trigger` to be listening to the `read_task`, but so far, the documentation is not very clear on how to exactly do that.

zhindes, 2022-03-09:
If you run NI I/O Trace while matlab does its thing, you should be able to see what driver operations they're doing. My guess is that it is sharing the start triggers like you suggested. Typically you would share the start trigger from AI to AO, start AO first, and then start and read on AI. [ai_multi_task_pxie_ref_clk.py](https://github.com/ni/nidaqmx-python/blob/master/nidaqmx_examples/ai_multi_task_pxie_ref_clk.py) does something similar to that, actually but across multiple devices. In your code, you would do something like this before you commit:
```
write_task.triggers.start_trigger.cfg_dig_edge_start_trig("/Dev2/ai/StartTrigger")
```

You can make it more generic like this, I believe:
```
write_task.triggers.start_trigger.cfg_dig_edge_start_trig(read_task.start_trigger.term)
```

If you get something working to your liking, I'd love to accept a contribution of a complete example from you!

fhchl, 2022-03-10:
Excellent! Your comment made me get it to work. My first tries replaced `write_task.start()` with something along the lines you recommended, but then the `write_task` wouldn't start at all. 

One needs to configure the trigger _and_ do `write_task.start()` to make it working:

```python
write_task.triggers.start_trigger.cfg_dig_edge_start_trig(read_task.triggers.start_trigger.term)
write_task.write(outdata, auto_start=False)
write_task.start()
indata = read_task.read(nsamples)
```

Committing the tasks explicitly doesn't seem to make any difference now, so I removed those lines. 

I will test the code a little more to see if I get exactly the same results as through Matlab and will gladly contribute it as an example. Thanks for the help 🙏🏼 !

ggggggggg, 2022-03-30:
Here's what I put together synthesizing the comments here:
```
import numpy as np
import nidaqmx as ni
from nidaqmx.constants import TaskMode
import pylab as plt

def query_devices():
    local_system = ni.system.System.local()
    driver_version = local_system.driver_version

    print('DAQmx {0}.{1}.{2}'.format(driver_version.major_version, driver_version.minor_version,
                                    driver_version.update_version))

    for device in local_system.devices:
        print('Device Name: {0}, Product Category: {1}, Product Type: {2}'.format(
            device.name, device.product_category, device.product_type))

def playrec(
    outdata, sr=96000, input_mapping=['Dev1/ai0'],
    output_mapping=['Dev1/ao0']
):
    nsamples = outdata.shape[0]
    with ni.Task() as read_task, ni.Task() as write_task:
        for o in output_mapping:
            aochan = write_task.ao_channels.add_ao_voltage_chan(o)
            aochan.ao_max = 3.5   # output range of USB-4431
            aochan.ao_min = -3.5
        for i in input_mapping:
            aichan = read_task.ai_channels.add_ai_voltage_chan(i)
            aichan.ai_min = -10
            aichan.ai_max = 10

        for task in (read_task, write_task):
            task.timing.cfg_samp_clk_timing(rate=sr, source='OnboardClock', samps_per_chan=nsamples)
        
        write_task.triggers.start_trigger.cfg_dig_edge_start_trig(read_task.triggers.start_trigger.term)
        write_task.write(outdata, auto_start=False)
        write_task.start()
        indata = read_task.read(nsamples)
        
    return indata

if __name__ == "__main__":
    query_devices()

    sr = 96000
    duration = 1
    t = np.linspace(0, duration, sr*duration, endpoint=False)
    sig = 1 * np.sin(1000*2*np.pi*t)

    indata = playrec(sig)
    plt.plot(t, indata)
    plt.plot(t, sig)
    plt.show()
```

shawnalbertson, 2022-08-11:
Thank you all for sharing your ideas, I've found them extremely helpful.

I'll add that the original source code has a series of [tests](https://github.com/ni/nidaqmx-python/tree/master/nidaqmx/tests) which illustrate some additional functionality, including using a [counter output pulse train task as a clock source for synchronous continuous read/ write operation](https://github.com/ni/nidaqmx-python/blob/master/nidaqmx/tests/test_stream_analog_readers_writers.py).

Seems like everyone got what they needed out of this issue, should we close it? @fhchl maybe upload a working example for good measure...

closed 2022-08-11 by zhindes

fhchl, 2022-08-12:
My code example would be similar to @ggggggggg 's above with two little extras:

- check that write data is not clipping the output
- do not time out for long measurements

```python
def playrec(
  data, samplerate=96000, input_mapping=['Dev2/ai0', 'Dev2/ai1', 'Dev2/ai2'],
  output_mapping=['Dev2/ao0']
):
  """Simultaneous playback and recording though NI device.

  Parameters:
  -----------
  data: array_like, shape (nsamples, len(output_mapping))
    Data to be send to output channels.  
  samplerate: int
    Samplerate
  input_mapping: list of str
    Input device channels
  output_mapping: list of str
    Output device channels

  Returns
  -------
  ndarray, shape (nsamples, len(input_mapping))
    Recorded data

  """
  max_out_range = 3.5 # output range of USB-4431
  max_in_range = 10   # input range of USB-4431
  max_outdata = np.max(np.abs(outdata))
  if max_outdata > max_out_range:
    raise ValueError(
      f"outdata amplitude ({max_outdata:.2f}) larger than allowed range"
      f"(+-{max_out_range}).")

  outdata = np.asarray(outdata).T
  nsamples = outdata.shape[1]

  with ni.Task() as read_task, ni.Task() as write_task:
    for o in output_mapping:
      aochan = write_task.ao_channels.add_ao_voltage_chan(o)
      aochan.ao_max = max_out_range
      aochan.ao_min = -max_out_range
    for i in input_mapping:
      aichan = read_task.ai_channels.add_ai_voltage_chan(i)
      aichan.ai_min = -max_in_range
      aichan.ai_max = max_in_range

    for task in (read_task, write_task):
      task.timing.cfg_samp_clk_timing(rate=samplerate, source='OnboardClock',
                                      samps_per_chan=nsamples)

    # trigger write_task as soon as read_task starts
    write_task.triggers.start_trigger.cfg_dig_edge_start_trig(
            read_task.triggers.start_trigger.term)
    write_task.write(outdata, auto_start=False)
    write_task.start()  # write_task doesn't start at read_task's start_trigger
                        # without this
    indata = read_task.read(nsamples, timeout=WAIT_INFINITELY) # do not time out
                                                               # for long inputs

  return np.asarray(indata).T
```

If this is useful I could open a PR for this example. Its form might be a little different from the style of the current examples, though.

zhindes, 2022-08-12:
Very nice example, I'd love to accept it!

ExoR24, 2023-10-13:
Please forgive my ignorance but I'm struggling through getting this example to work (I'm new to interfacing with NI equipment), and the example from the PR. I'm working with a cDAQ-9174 and trying to play record through cards 9264 and 9234. I keep getting the error that I can 't get read_task.triggers.start_trigger.term until the task is running.

Specifically the error:

```
DaqError: You only can get the specified property while the task is reserved, committed or while the task is running.
Reserve, commit or start the task prior to getting the property.
Property: DAQmx_StartTrig_Term

Task Name: _unnamedTask<65>

Status Code: -200983
```

Any ideas what I'm doing wrong?

bkeryan, 2023-10-13:
The cDAQ-9174 has three analog input timing engines, and each timing engine has a different start trigger terminal. Before you can query the start trigger terminal, you need to reserve `read_task` so that DAQmx chooses which timing engine to use.

Try calling `read_task.control(TaskMode.TASK_RESERVE)` above the call to `write_task.triggers.start_trigger.cfg_dig_edge_start_trig`.

ExoR24, 2023-10-13:
Awesome thanks @bkeryan! That worked for me