ni/nidaqmx-python #163: Nidaqmx-python interferes with labview when a Dev is in use in each

tozmann, 2022-03-22:
I have two usb-6002. I use one (dev1) with a Labview (vi) executable and the second one (dev2) with nidaqmx-python. I have the executable running constantly (analog out) and I run the python code when needed only (analog in). The problem is that the sampling rate in dev1 changes when I execute the python code that is supposed to be talking to dev2 only. In the python code I create a task, set a sampling rate and read voltages. The change to dev1 is not immediate. It seems to occur towards the end of the python code execution. As if some default parameters where written back to all possible channels, without discriminating the Devs

zhindes, 2022-03-23:
Can you share a screenshot of the LV code and the Python source?

tozmann, 2022-03-23:
Sure. Here you go! 
[image]
[image]

zhindes, 2022-03-23:
What do you mean by "The problem is that the sampling rate in dev1 changes when I execute the python code that is supposed to be talking to dev2 only. ", specifically? In your VI, `Dev3/ai0:1` is hardware-timed, so it shouldn't change sampling rate at all. However, `Dev3/ao0:1` and `Dev3/port1/line1` are software-timed. I imagine what is occurring is that when your Python program finishes the timed loop is slowing down for a few iterations causing the AO to slow down. One way to avoid that would be to:
* Configure a Sample Clock on the AO task
* Write your buffer once before you start
* Let the AO continue to output with hardware timing throughout the loop. You should only need to call `DAQmx Is Task Done` to see if it errored.

tozmann, 2022-03-23:
What happens is that the speed or sampling rate or something in AO of dev3
changes when the python code finishes. It doesn't slow down for a bit. I am
outputting my (12 points) data in a loop, where each interation is meant to
last 12 ms, and I do this non stop. I then run the python code, and once it
finishes, this 12 ms turn into something like 60 ms.

To have the AO be 12 ms again, I have to stop the VI and restart it.
Otherwise it stays at that other period forever.

If instead of the python code, I do a VI with the exact same functionality,
then all is fine: the 12 ms stay at 12 ms.

That's why I say that I could be that the closing of task (dev1) in the
python code is changing something in the dev3 back to a default or
something.

I'll try your suggestions and let you know.

zhindes, 2022-03-23:
Interesting. Does this happen if you set the timer to 30 ms? Does it get slower? The Windows timer resolution is typically ~16ms, but it can be changed and often is. See [Windows Timer Resolution: Megawatts Wasted](https://randomascii.wordpress.com/2013/07/08/windows-timer-resolution-megawatts-wasted/). I suspect the LV timed loop may be doing that. Try out [clockres](https://docs.microsoft.com/en-us/sysinternals/downloads/clockres) to see if it changing when you see the loop run slower. Maybe something DAQ or Python is doing is slowing it down? Regardless, making the AO task HW timed will eliminate that as a factor for the signal you're generating.

tozmann, 2022-03-23:
I have a usb-6002 as daq an apparently they cannot do HW timing :-/. Cheap
things are expensive.

I will look into Window's clock and let you know. Thanks for all the
suggestions BTW!

tozmann, 2022-03-24:
Hi,

So I started monitoring Windows' clock, and since then I haven't seen the
same behaviour. It now behaves how it should.

I also updated my conda and I was checking the different SW clocks I have
available in the VI's for (timed) loop.

Not sure what could have fixed the problem. I cannot reproduce it anymore
😅. Thanks a lot for your help. Hopefully it will stay like this.

Best
T

zhindes, 2022-03-24:
Hmm, odd! I suspect that _something_ is mucking with the Windows clocks. It is a pretty common problem, that not many applications will notice.

> I have a usb-6002 as daq an apparently they cannot do HW timing :-/.

Weird, I looked that up and it says it does per the [specifications](https://www.ni.com/pdf/manuals/374371a.pdf).

[image]

> Thanks a lot for your help. Hopefully it will stay like this.

I hope so, too!

closed 2022-03-24 by zhindes