ni/nidaqmx-python #164: Numpy version requirement conflicts with Anaconda

edbarnard, 2022-03-29:
The current pip-installable version nidaqmx-python (nidaqmx-0.6.1-py3-none-any.whl) requires numpy<2.0,>=1.22

However the latest anaconda distribution (2021.11) is currently locked at numpy 1.20. During installation, pip will try to upgrade numpy to 1.22-3, but this causes numerous conflicts for other default anaconda packages:

> ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
> numba 0.54.1 requires numpy<1.21,>=1.17, but you have numpy 1.22.3 which is incompatible.

This makes it currently impossible to use nidaqmx with the standard Anaconda distribution.

Is there any technical reason for the numpy>=1.22 requirement? If not, I would suggest that the minimum numpy version should be reduced to at least numpy-1.20

zhindes, 2022-03-29:
numpy changed their Python requirements recently.
* In [1.21.5](https://pypi.org/project/numpy/1.21.5/): Requires: Python >=3.7, <3.11 
* In [1.22.0+](https://pypi.org/project/numpy/1.22.0/): Requires: Python >=3.8 (they dropped 3.7).

Since we still support Python 3.7, we had to lock down numpy as follows:
```toml
numpy = [
    {version="1.21.5", python = "~3.7"},
    {version="^1.22", python = "^3.8"}
]
```

However, it looks like numpy 1.20.0 also has Python >=3.7. So I think we could change that numpy requirement to this, and it would still work:
```toml
numpy = [
    {version="1.21.5", python = "~3.7"},
    {version="^1.20", python = "^3.8"}
]
```

If you want to try that out these are the steps:
* Update pyproject.toml
* Run `poetry update`
* Run tests on all supported Python versions (3.7-3.10), see [CONTRIBUTING](https://github.com/ni/nidaqmx-python/blob/master/CONTRIBUTING.md).

I'd be happy to take that contribution if it works!

pieleric, 2023-08-08:
Hi,
I have somehow a similar situation here. We'd like to use nidaqmx on Ubuntu 20.04. It has Python 3.8 and numpy 1.17. Does nidaqmx-python require any special feature/function introduced in numpy after v1.17? I've tried it (by changing the minimum numpy version required), and my simple tests seemed to work fine.

In general, why would *you* need to care about which Python version is supported by numpy? In the requirements for nidaqmx-python should be only stated which minimum version this code needs. Then dependency manager (like conda or pip) will take care of installing the right numpy given the Python version and requirements from nidadmx-python...

zhindes, 2023-08-08:
Its a long story, but the summary is that this project was abandoned for a long time. When I picked it up, I switched to poetry, and had to fix all the ancient dependency resolution issues. I picked the latest version of numpy that was available at the time, and had to deal with the Python 3.7 vs 3.8+ issue. At this point, we can drop Python 3.7 since it is end-of-life, and I'm happy to take a contribution that cleans up the numpy dependency to be something a bit older as well.

bkeryan, 2023-08-08:
> In general, why would you need to care about which Python version is supported by numpy?

I think this is also related to how Poetry dependency locking works.

When you run `poetry lock`, Poetry downloads info from PyPI about what packages are available and chooses the newest set of packages that matches the version constraints in the `pyproject.toml` file.

If we specified `numpy = ">=1.20"` (or an older version like 1.17), Poetry would try to find a single version of `numpy` that works with Python >= 3.7.
- `numpy` 1.22 dropped support for Python 3.7, so Poetry would have to choose an older version.
- `numpy` 1.21.6 requires Python >=3.7, <3.11,
- `numpy` 1.20 requires Python >= 3.7, but it doesn't have binary wheels for Python >= 3.10. Installing this version will try to rebuild NumPy from source.

Also, `numpy` has no dependencies on other packages, which triggers a legacy quirk in PyPI metadata handling. My understand is that when the PyPI metadata for a package doesn't list any dependencies, it is ambiguous whether a package has no dependencies or whether the package has dependencies that are specified in `setup.py` but not in PyPI's metadata. This situation causes dependency resolvers like Poetry and `pip` to backtrack, downloading multiple versions of `numpy` until they find a version that meets the constraints. I remember this taking somewhere between several minutes and half an hour. For more details, see: [Poetry FAQ: Why is the dependency resolution process slow?](https://python-poetry.org/docs/faq/#why-is-the-dependency-resolution-process-slow), [pip 20.3.2 eagerly downloads all versions of a package during dependency resolution #9284](https://www.github.com/pypa/pip/issues/9284)

I think that maximizing compatibility while avoiding backtracking is possible. It probably involves specifying a `numpy` version constraint for each supported Python version, based on the oldest version of `numpy` that supported that Python version.

pieleric, 2023-08-09:
Thanks for the clarification. Do I conclude correctly that at least numpy v1.20 is supported. Maybe even earlier versions work but you haven't checked? If so, there could be chances that it indeed works on v1.17 as my preliminary tests showed?

zhindes, 2023-08-09:
Yes, it almost certainly works. If you are able to fully run the tests, then I am confident.

zhindes, 2024-01-18:
Closing due to inactivity. If somebody wants to contribute a fix, they can.

closed 2024-01-18 by zhindes