ni/nidaqmx-python #165: ipython task.timing.cfg_samp_clk? help fails after task started

ggggggggg, 2022-03-30:
```
Python 3.8.10 (default, Mar 15 2022, 12:22:08) 
Type 'copyright', 'credits' or 'license' for more information
IPython 7.13.0 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import nidaqmx as ni 
   ...: import nidaqmx.stream_readers 
   ...: import nidaqmx.constants 
   ...: task = ni.Task() 
   ...: task.ai_channels.add_ai_voltage_chan('Dev1/ai0') 
   ...: task.timing.cfg_samp_clk_timing(rate=10000, source="OnboardClock",  
   ...: sample_mode=nidaqmx.constants.AcquisitionType.CONTINUOUS) 
   ...: reader =  ni.stream_readers.AnalogMultiChannelReader(task.in_stream) 
   ...: task.start() 
   ...:                                                                         

In [2]: task.timing.cfg_samp_clk_timing?                                        
/usr/lib/python3.8/subprocess.py:946: ResourceWarning: subprocess 265146 is still running
  _warn("subprocess %s is still running" % self.pid,
ResourceWarning: Enable tracemalloc to get the object allocation traceback
```
Using the help works at any point before calling `task.start()` in the above example. It should keep working after as well.

ggggggggg, 2022-03-31:
It doesn't seem like asking for help should be calling out to subprocess, just looking up something in a dict.

bkeryan, 2024-01-18:
This may be related to the fact that `nidaqmx` changes the global warning filter to enable `ResourceWarning`: #325