ni/nidaqmx-python #166: add overview to documentation

ggggggggg, 2022-04-15:
The documentation basically just documents each function... but it's hard to understand without some sort of high level overview. Like how the timing system works, what a task is, the limitations on having separate tasks, etc. I assume this stuff in the c documentation? So far I haven't found it.

zhindes, 2022-04-18:
Does [Getting Started with NI-DAQmx](https://www.ni.com/en-us/support/documentation/supplemental/06/getting-started-with-ni-daqmx--main-page.html) help you find what you're looking for? I'm happy to accept a contribution that updates the documentation to link out to that!

ggggggggg, 2022-04-20:
That definitely helps, I can try making a PR in a few days,.

zhindes, 2024-01-18:
Closing due to inactivity. If someone wants to pick up #171, feel free.

closed 2024-01-18 by zhindes