ni/nidaqmx-python #181: mismatched enum names in ai_term_cfgs

paulcdca, 2022-07-15:
This was found following the '_example of using an nidaqmx.system.System object_' at [https://nidaqmx-python.readthedocs.io/en/latest/](https://nidaqmx-python.readthedocs.io/en/latest/)

```
import nidaqmx
nidaqmx.system.PhysicalChannel('Dev1/ai0').ai_term_cfgs

Traceback (most recent call last):
  File "<input>", line 1, in <module>
  File "D:\RobotFrameworkTrial\venv\lib\site-packages\nidaqmx\system\physical_channel.py", line 402, in ai_term_cfgs
    return enum_bitfield_to_list(
  File "D:\RobotFrameworkTrial\venv\lib\site-packages\nidaqmx\_lib.py", line 93, in enum_bitfield_to_list
    enum_value = next(
StopIteration
```

Looking under the hood, this exception is caused because these two enums in constants.py dont share the same names and so enum_bitfield_to_list calls next with an empty list

```
class TerminalConfiguration(Enum):
    DEFAULT = -1  #: Default.
    RSE = 10083  #: Referenced Single-Ended.
    NRSE = 10078  #: Non-Referenced Single-Ended.
    BAL_DIFF = 10106  #: Differential.
    PSEUDODIFFERENTIAL = 12529  #: Pseudodifferential.

class _TermCfg(Enum):
    RSE = 1  #: RSE terminal configuration
    NRSE = 2  #: NRSE terminal configuration
    DIFF = 4  #: Differential terminal configuration
    PSEUDO_DIFF = 8  #: Pseudodifferential terminal configuration

```

zhindes, 2022-07-15:
D'oh, nice find. This is also broken for `device.ai_couplings` because of these two enums:

```python
class _CouplingTypes(Enum):
    AC = 1  #: Device supports AC coupling
    DC = 2  #: Device supports DC coupling
    GROUND = 4  #: Device supports ground coupling
    HF_REJECT = 8  #: Device supports High Frequency Reject coupling
    LF_REJECT = 16  #: Device supports Low Frequency Reject coupling
    NOISE_REJECT = 32  #: Device supports Noise Reject coupling

class Coupling(Enum):
    AC = 10045  #: Remove the DC offset from the signal.
    DC = 10050  #: Allow NI-DAQmx to measure all of the signal.
    GND = 10066  #: Remove the signal from the measurement and measure only ground.
```

I've got some spare time today, so I'll look into why our internal metadata/generator is being inconsistent here.

zhindes, 2022-07-18:
Fixing here #182

closed 2022-07-19 by zhindes