ni/nidaqmx-python #188: Expose Shunt Calibration and Offset Nulling Calibration Functions in Bindings

jeffreylangston29, 2022-09-02:
[image]

Requesting support for the equivalent to the above portion of the LabVIEW shipping example **_Bridge - Continuous Input.vi_** be exposed in the nidaqmx-python bindings. The above example shows the bridge calibration functions: **_DAQmx Perform Shunt Calibration.vi_** and **_DAQmx Perform Bridge Offset Nulling Calibration.vi_**.

Exposing bindings for bridge, strain, and thermocouple lead functions would be ideal.

It appears that someone else is interested in this functionality as well:

[https://forums.ni.com/t5/Multifunction-DAQ/NiDaqmx-Python-Perform-Bridge-Offset-Nulling-Calibration/td-p/4166367](url)

Based on some cursory digging, the necessary functions are exported in the c library:

[image]

zhindes, 2022-09-19:
We aren't planning on adding device calibration support, and these entrypoints were caught up in that decision. I agree that they should be added. There isn't any plan to do that in the near-term, so if you wanted to contribute - feel free!

nicjohnston, 2023-03-07:
Is there any chance this has changed in the last 6 months?  I'm working on a cDAQ system with strain gauges, the offset nulling and shunt calibration are critical features for us.

zhindes, 2023-03-08:
Sorry, no change. However, you could do this yourself, if you wanted. Under the hood we use ctypes to load the nidaqmx DLL and call functions on it. You can see how we load it in [_import_lib](https://github.com/ni/nidaqmx-python/blob/master/src/nidaqmx/_lib.py#L182). We call functions directly on that ctypes object. For example, [here](https://github.com/ni/nidaqmx-python/blob/master/src/nidaqmx/_task_modules/ai_channel_collection.py#L122) is where we call `DAQmxCreateAIAccel4WireDCVoltageChan` on the DLL. You can see that we define the function types and call it.

```python
cfunc = lib_importer.windll.DAQmxCreateAIAccel4WireDCVoltageChan
if cfunc.argtypes is None:
    with cfunc.arglock:
        if cfunc.argtypes is None:
            cfunc.argtypes = [
                lib_importer.task_handle, ctypes_byte_str,
                ctypes_byte_str, ctypes.c_int, ctypes.c_double,
                ctypes.c_double, ctypes.c_int, ctypes.c_double,
                ctypes.c_int, ctypes.c_int, ctypes.c_double, c_bool32,
                ctypes_byte_str]

error_code = cfunc(
    self._handle, physical_channel, name_to_assign_to_channel,
    terminal_config.value, min_val, max_val, units.value, sensitivity,
    sensitivity_units.value, voltage_excit_source.value,
    voltage_excit_val, use_excit_for_scaling, custom_scale_name)
```

For the functions you care about, you can look at the `NIDAQmx.h` header file for their signature.
```c
int32 __CFUNC     DAQmxPerformBridgeOffsetNullingCal(TaskHandle taskHandle, const char channel[]);
int32 __CFUNC     DAQmxPerformBridgeOffsetNullingCalEx(TaskHandle taskHandle, const char channel[], bool32 skipUnsupportedChannels);
int32 __CFUNC     DAQmxPerformThrmcplLeadOffsetNullingCal(TaskHandle taskHandle, const char channel[], bool32 skipUnsupportedChannels);
// Note: This function is deprecated.  Use DAQmxPerformStrainShuntCalEx instead.
int32 __CFUNC     DAQmxPerformStrainShuntCal     (TaskHandle taskHandle, const char channel[], float64 shuntResistorValue, int32 shuntResistorLocation, bool32 skipUnsupportedChannels);
int32 __CFUNC     DAQmxPerformStrainShuntCalEx   (TaskHandle taskHandle, const char channel[], float64 shuntResistorValue, int32 shuntResistorLocation, int32 shuntResistorSelect, int32 shuntResistorSource, bool32 skipUnsupportedChannels);
// Note: This function is deprecated.  Use DAQmxPerformBridgeShuntCalEx instead.
int32 __CFUNC     DAQmxPerformBridgeShuntCal     (TaskHandle taskHandle, const char channel[], float64 shuntResistorValue, int32 shuntResistorLocation, float64 bridgeResistance, bool32 skipUnsupportedChannels);
int32 __CFUNC     DAQmxPerformBridgeShuntCalEx   (TaskHandle taskHandle, const char channel[], float64 shuntResistorValue, int32 shuntResistorLocation, int32 shuntResistorSelect, int32 shuntResistorSource, float64 bridgeResistance, bool32 skipUnsupportedChannels);
```