ni/nidaqmx-python #189: in_Stream overtemperature_chans return -201194 although overtemperature_chans_exist had been called

archerleong, 2022-09-19:
**Hardware setup**
TS-15010, TS 15050 and TS-15200

**Details**
Although overtemperature_chans_exist had been called before overtemperature_chans , it still assume that overtemperature_chans_exist is not called and return -201194. Similar behavior happened to other properties as well such as reverse_voltage_error_chans, remote_sense_error_chans, aux_power_error_chans, etc. Any properties which required xxxxx_chans_exist to be called prior will be facing similar issue.

**Recommendation** 
The issue can be fixed by assigning expected buffer size. 
Recommended solutions:-
1.  exposing buffer size as argument and allow user to decide the buffer size.
2.  allocating bigger buffer based on the number of channels in the task.

charitylxy, 2024-03-07:
Hi @archerleong, I have a few questions regarding this as I can't seems to reproduce the error. 
- Does this issue only happens on real device? I'm testing it on simulated device and it's seems to be working. 
- What are the steps I can make to cause the module to overtemp? I'm not too familiar with TestScale, currently I'm just overriding the `overtemperature_chans_exist` property to `True` to test it out. 

Appreciate your inputs on this, thanks!

archerleong, 2024-03-07:
Yes. It required actual TS15200 device. 

"What are the steps I can make to cause the module to overtemp? I'm not too familiar with TestScale, currently I'm just overriding the overtemperature_chans_exist property to return True to test it out."
You can reduce the temperature limit threshold. I forwarded you the test case in Teams, you may look at the VIs on how to do that.

bkeryan, 2024-03-07:
@archerleong @charitylxy This also affects grpc-device and other C applications. I filed a DAQmx bug about it last May: AB#2393824: DAQmx read/write status properties return errors when called from C, Python, or grpc-device

charitylxy, 2024-03-12:
@bkeryan This issue is caused by [AB#2393824](https://dev.azure.com/ni/DevCentral/_workitems/edit/2393824) you mentioned, as the first time calling `xxx_chans` after `xxx_chans_exist` will return the actual buffer size needed since it's set as 0 initially [here](https://github.com/ni/nidaqmx-python/blob/0f65354421c5de8799eef8e1343db6f16af95524/generated/nidaqmx/_library_interpreter.py#L3241). Which means `xxx_chans` will need to be called another time which will then return error -201194.

I have a few ideas on fixing this, but facing some road blocks:
1. The ideal would be fixing [AB#2393824](https://dev.azure.com/ni/DevCentral/_workitems/edit/2393824) to allow multiple `xxx_chans` calls after `xxx_chans_exist` is being called once but this might requires big effort as it involves various family-specific AI/AO/DIO experts. 
2. Another temporary workaround would be getting the channel names to get the buffer size at `xxx_chans` property and passing the buffer size to `get_read_attribute_string` at `_library_interpreter` however this means we'll probably have to make these handwritten since  these are all codegened, not sure if that's appropriate. 

Perhaps there's a better way in fixing this, appreciate your thoughts on this.

bkeryan, 2024-03-12:
> 2\. Another temporary workaround would be getting the channel names to get the buffer size at `xxx_chans` property and passing the buffer size to `get_read_attribute_string` or `get_write_attribute_string` at `_library_interpreter` however this means we'll probably have to make these handwritten since  these are all codegened, not sure if that's appropriate.

You could hack the codegen to do this for all read/write status string attributes.

closed 2024-03-14 by charitylxy