ni/nidaqmx-python #199: `OSError` when using nidaqmx on Windows: "initialization routine failed"

RobertoRoos, 2023-01-05:
I am trying out `nidaqmx` from Windows Sandbox. I boot up a new sandbox and install NI DAQmx, without all the optional features.

If I then install Python 3.10, create a venv, install `nidaqmx` and run for example "examples/system_properties.py", I get the following error:

> Traceback (most recent call last):
>   File "C:\Users\WDAGUtilityAccount\Downloads\nidaqmx-python\nidaqmx-python-master\nidaqmx_examples\system_properties.py", line 5, in <module>
>     driver_version = local_system.driver_version
>   File "C:\Users\WDAGUtilityAccount\Downloads\nidaqmx-python\nidaqmx-python-master\venv\lib\site-packages\nidaqmx\system\system.py", line 71, in driver_version
>     return System.DriverVersion(self._major_version, self._minor_version,
>   File "C:\Users\WDAGUtilityAccount\Downloads\nidaqmx-python\nidaqmx-python-master\venv\lib\site-packages\nidaqmx\system\system.py", line 106, in _major_version
>     cfunc = lib_importer.windll.DAQmxGetSysNIDAQMajorVersion
>   File "C:\Users\WDAGUtilityAccount\Downloads\nidaqmx-python\nidaqmx-python-master\venv\lib\site-packages\nidaqmx\_lib.py", line 161, in windll
>     self._import_lib()
>   File "C:\Users\WDAGUtilityAccount\Downloads\nidaqmx-python\nidaqmx-python-master\venv\lib\site-packages\nidaqmx\_lib.py", line 204, in _import_lib
>     windll = ctypes.windll.LoadLibrary(lib_name)
>   File "C:\Program Files\Python310\lib\ctypes\__init__.py", line 452, in LoadLibrary
>     return self._dlltype(name)
>   File "C:\Program Files\Python310\lib\ctypes\__init__.py", line 374, in __init__
>     self._handle = _dlopen(self._name, mode)
> OSError: [WinError 1114] A dynamic link library (DLL) initialization routine failed

RobertoRoos, 2023-01-05:
I ran [Dependencies](https://lucasg.github.io/Dependencies/) in the sandbox and it can load `nicaiu.dll` without problems. Only it cannot locate `NiSpyLog.dll`, but on my real system Dependencies gives the same output and I can use `nidaqmx` without problems there.

zhindes, 2023-01-05:
FYI - while Windows Sandbox still isn't officially supported by NI drivers, we did have a developer try it out with the following setup and be successful:
* NI-DAQmx 22.8
* Windows 11 22H2 - note that some new functionality was added to sandbox in this release. In particular, reboots are now supported which our services require after installation.

RobertoRoos, 2023-01-06:
Windows 11 could be the difference here. Maybe the reboot alone is what's important.

Thanks for your input, I guess there's not much to be done about this.

closed 2023-01-06 by RobertoRoos

RobertoRoos, 2023-01-06:
Interesting, tried it from scratch and now it works!

I installed DAQmx, this time installing all the default components instead of deselecting all the optional stuff and I run the scripts normally.