ni/nidaqmx-python #200: Problem with PyInstaller

kasasxav, 2023-01-16:
Hi,

I am experiencing an error when I try to compile my project with pyinstaller, with the nidaqmx depencency.

It runs this error:
importlib.metadata.PackageNotFoundError: nidaqmx

After some search I found this: https://stackoverflow.com/questions/71226142/pyinstaller-with-nidaqmx
It seems to work, but I was wondering if you would consider adding this into the main release, since I compile my project through Github Actions and modifyign the nidaqmx is only possible in my local computer.

thanks!

zhindes, 2023-01-17:
From that Stack Overflow thread, it looks like there is a solution that doesn't require modifying the source using the [copy-metadata command line option of PyInstaller](https://pyinstaller.org/en/stable/usage.html#cmdoption-copy-metadata). Does that work for you?

zhindes, 2023-01-17:
FYI - providing a version is a useful paradigm of most modules, see [Single-sourcing the package version](https://packaging.python.org/en/latest/guides/single-sourcing-package-version/). I don't want to remove that functionality here. As noted in that guide, the metadata approach is a standard one, so I think what we have is fine. Hopefully, the `pyinstaller` workaround is sufficient.