ni/nidaqmx-python #201: Starting a Task with AnalogIn channels changes voltage

RobertoRoos, 2023-01-19:
I found out that starting a task and doing a `read()` causes a tiny change in the measured voltage.

I am making an application where I'm doing slow and long measurements at 1 sample per second. My approach was like:
```
def init():
    self.device = ...
    self.task = Task()
    self.task.ai_channels.add_ai_voltage_chan("Dev0/ai0:3")

def sample():
    result = self.task.read() 

# Timer for `sample`
```

I.e. I create a task and run it every second. I got some tiny discrepancies that my colleague did not get when using DAQExpress, which confused me.
With an external scope I saw that I got a tiny voltage of about 50 mV whenever `task.read()` was called.

Then I noticed I only get this on the first sample. When taking multiple samples in a row every second:
```
def sample():
    results = self.task.read(number_samples_per_channel=10)
```
...The first few values are distorted but the last ones are as expected.

Then if I put the task on it's own timer, this distortion isn't there at all:
```
def init():
    self.device = ...
    self.task = Task()
    self.task.ai_channels.add_ai_voltage_chan("Dev0/ai0:3")
    self.task.timing.cfg_samp_clk_timing(1, sample_mode=AcquisitionType.CONTINUOUS)
    self.task.register_every_n_samples_acquired_into_buffer_event(1, sample)

def sample():
    result = self.task.read(1)
    return 0

# No other timers
```

My guess is that if the sample mode is not `CONTINUOUS`, the input impedance changes during `.read()` and changes back after, causing a fluctuation because there is capacitance in my measured system.

Is this something by design?

I would prefer to time every `.read()` from an external timer, instead of relying on buffers. But this distortion is not acceptable.

Thank you for any insight!

RobertoRoos, 2023-01-19:
To clarify, these discrepancies in voltage (a thin peak of about 50 mV) is measured both by the DAQ as an external probe.

RobertoRoos, 2023-01-19:
I extracted a real minimal demonstration from my application.

So this shows a problem:
```
import nidaqmx
import nidaqmx.system
from time import sleep

system = nidaqmx.system.System.local()
device = system.devices[0]

while True:
    with nidaqmx.Task() as task:
        task.ai_channels.add_ai_voltage_chan(f"{device.name}/ai0:3")
        values = task.read(number_of_samples_per_channel=5)

    print(values)

    sleep(0.5)
```
On each measurement the samples show a drift:

> ...
> [[1.4308285370934755, 1.4372671274468303, 1.415375920245424, 1.4050741756800562, 1.4012110214680433, 1.3986355853267014, 1.3960601491853595, 1.3934847130440176, 1.3921969949733466, 1.3909092769026756], ... ]
> [[1.4346916913054883, 1.4398425635881722, 1.4179513563867658, 1.408937329892069, 1.4037864576093853, 1.4012110214680433, 1.3986355853267014, 1.3973478672560304, 1.3947724311146885, 1.3934847130440176], ... ]
> [[1.4334039732348174, 1.4385548455175012, 1.4192390744574368, 1.4063618937507272, 1.4037864576093853, 1.4012110214680433, 1.3999233033973724, 1.3973478672560304, 1.3960601491853595, 1.3947724311146885], ... ]
> ...

While this works fine:
```
import nidaqmx
import nidaqmx.system
from nidaqmx.constants import AcquisitionType, READ_ALL_AVAILABLE

system = nidaqmx.system.System.local()
device = system.devices[0]

with nidaqmx.Task() as task:
    task.ai_channels.add_ai_voltage_chan(f"{device.name}/ai0:3")
    task.timing.cfg_samp_clk_timing(20, sample_mode=AcquisitionType.CONTINUOUS)

    def callback(*_):
        values = task.read(number_of_samples_per_channel=READ_ALL_AVAILABLE)
        print(values)
        return 0

    task.register_every_n_samples_acquired_into_buffer_event(10, callback)

    task.start()

    while True:
        pass  # Hang
```

Now the measurements are constant:

> ...
> [[1.4424179997295141, 1.443705717800185, 1.443705717800185, 1.443705717800185, 1.4424179997295141, 1.4424179997295141, 1.4424179997295141, 1.443705717800185, 1.443705717800185, 1.443705717800185], ... ]
> [[1.443705717800185, 1.443705717800185, 1.443705717800185, 1.4424179997295141, 1.443705717800185, 1.443705717800185, 1.443705717800185, 1.4424179997295141, 1.443705717800185, 1.443705717800185], ... ]
> [[1.443705717800185, 1.443705717800185, 1.444993435870856, 1.443705717800185, 1.443705717800185, 1.443705717800185, 1.443705717800185, 1.443705717800185, 1.443705717800185, 1.443705717800185], ... ]
> ...

And I would expect identical results.

zhindes, 2023-01-19:
In my experience with DAQ devices those slight fluctuations in measured voltage at a beginning of a scan is from settling of switching the front end configuration. There is a single ADC and muxes to switch the various inputs and terminal configurations.

For example, `add_ai_voltage_chan` has a parameter, `terminal_config=TerminalConfiguration.DEFAULT`, that is set to some default value based on the device. It looks like the 6001 supports Differential on the first 4 channels, and RSE on all 8 channels. For your application, then, it is probably using Differential. I am guessing as a task starts the front end is switching from some default state to that differential configuration. If you switched to RSE, as defined [here](https://github.com/ni/nidaqmx-python/blob/f75a1c13fe5c437700e1066649de01f9e0b80d79/nidaqmx/constants.py#L832), it might not happen. Note that the terminal configuration is dependent on how you've connected your voltage signal to the device, so you may not be able to change it without rewiring.

All that being said, this behavior you're seeing is a device-specific issue, and not something specific to the Python support we provide here. The Python wrapper wouldn't induce this type of behavior. The [Multifunction DAQ Forum on ni.com](https://forums.ni.com/t5/Multifunction-DAQ/bd-p/250) is a great place to ask this type of question.

RobertoRoos, 2023-01-24:
Thanks for the suggestion.
For future reference, I posted on the NI forum too: https://forums.ni.com/t5/Multifunction-DAQ/USB-6001-different-results-in-continuous-and-finite-acquisition/m-p/4279380