ni/nidaqmx-python #216: Can read channel_names of PersistedTask but not channels

robin-po, 2023-03-02:
I created an analog output task in MAX named "AO" and 2 channels inside this task called "EV01" and "EV02".

I tried to launch this basic code :

```
import nidaqmx

task: nidaqmx.Task = nidaqmx.system.storage.persisted_task.PersistedTask('AO').load()

try:
    print(task.channel_names)
    print(task.channels)
finally:
    task.close()
```

Output :

```
['EV01', 'EV02']
Traceback (most recent call last):
  File "...\for_test\daq_test.py", line 7, in <module>
    print(task.channels)
          ^^^^^^^^^^^^^
  File "...\venv\Lib\site-packages\nidaqmx\task.py", line 163, in channels
    return Channel._factory(
           ^^^^^^^^^^^^^^^^^
  File "...\venv\Lib\site-packages\nidaqmx\_task_modules\channels\channel.py", line 115, in _factory
    check_for_error(error_code)
  File "...\venv\Lib\site-packages\nidaqmx\errors.py", line 196, in check_for_error
    raise DaqError(error_buffer.value.decode('iso-8859-1'), error_code)
nidaqmx.errors.DaqError: Specified channel is not in the task.
Channel Name: EV1

Task Name: AO

Status Code: -200486
```

It seems that I can read chanel_names but on reading channels it omits the "0"

Python version : 3.11.0 (main, Oct 24 2022, 18:26:48) [MSC v.1933 64 bit (AMD64)]
NI-DAQmx version : 2022 Q3

zhindes, 2023-03-02:
Oof, good catch. Sorry about that. The problem is that `flatten_channel_string` doesn't handle leading `0` in a channel name correctly.

```python
from nidaqmx.utils import flatten_channel_string

chans = ["EV01", "EV02"]
print(f"{chans} flattened: {flatten_channel_string(chans)}")
```

That code produces this incorrect output:

```
['EV01', 'EV02'] flattened: EV1:2
```

zhindes, 2023-03-02:
@robin-po - The workaround would be to rename your channels. The fix is to improve our `flatten_channel_string` implementation. I'm not sure when I'll get to that, but if you wanted to give it a shot (and add some tests!), I would appreciate the contribution :)

zhindes, 2023-03-02:
Got bored. Tentative fix here: https://github.com/zhindes/nidaqmx-python/tree/fix_flatten_unflatten_leading_zeros. Need to add more test cases and then validate all the other tests still work.

robin-po, 2023-03-03:
The fix works pretty well on my specific names, thanks !

PR #217 (2023-03-07): Handle leading zeros in flatten/unflatten implementation
We should now correctly handle leading zeros in channel names during flatten and unflatten. I also cleaned up the tests and documented the fact that our Python-only impl is not fully compatible with the NI-DAQmx driver.

Closes #216
closed 2023-03-07 by zhindes