ni/nidaqmx-python #218: empty list when we try to get device names (ubuntu 20.04)

csanrod, 2023-03-07:
We tried using nidaqmx library by using python3 with NI-USB 6003 but we get an empty list. Our system and drivers info are:

- Ubuntu 20.04 LTS 
- NILinux2023Q1DeviceDrivers (stream version)
- ni-daqmx (apt install) 
- NI-USB 6003
- nidaqmx (pip3 install)

We need to obtain devices names to assign physicalname channels like "Dev0/ai1:4". Here is what we obtain:

[image]

And here we can see that the device is detected:
[image]

Thanks for your help, best regards.

zhindes, 2023-03-07:
Unfortunately, the USB-6003 isn't supported on Linux. You can look up support for any given device on [NI Hardware and Software Operating System Compatibility (ni.com)](https://www.ni.com/en-us/support/documentation/compatibility/21/ni-hardware-and-operating-system-compatibility.html). That site is sort of hard to use, I apologize. Here is what it currently says.

[image]

A quick and easy way to find out if NI-DAQmx knows about your device is to run `nilsdev` on the command line. That should list all the devices that have been discovered, and it uses the same mechanism that the Python bindings use.

closed 2023-03-07 by zhindes

csanrod, 2023-03-08:
Oh ok, thanks for the information!