ni/nidaqmx-python #290: [WinError 1114] A dynamic link library (DLL) initialization routine failed

ChrisJones79, 2023-04-20:
## Python Version:
3.11.2 (tags/v3.11.2:878ead1, Feb  7 2023, 16:38:35) [MSC v.1934 64 bit (AMD64)]

## OS
Windows 10 Enterprise 22H2 (19045.2728)
Windows Feature Experience Pack 120.2212.4190.0

## CPU
12th Gen Intel(R) Core(TM) i5-1245U   1.60 GHz
16.0 GB (15.7 GB usable)
64-bit operating system, x64-based processor

## Driver (Installed with NI Package Manager) and Device
USB-6002
USB\VID_3923&PID_76C4&REV_0100
USB\VID_3923&PID_76C4
23.3.0.49311
nixfmrrkw
oem384.inf
oem384.inf:USB\VID_3923&PID_76C4&REV_0100,USB-6002.Cfg
{Unable to Load Device Driver}
%hs device driver could not be loaded.
Error Status was 0x%x
USB\Class_FF&SubClass_00&Prot_00
USB\Class_FF&SubClass_00
USB\Class_FF
C000026C

NI-DAQmx	National Instruments	Drivers	2023 Q2	 
NI-DAQmx .NET Language Runtime for .NET Framework 4.0	National Instruments	Drivers	2023 Q2	 
NI-DAQmx .NET Language Runtime for .NET Framework 4.5	National Instruments	Drivers	2023 Q2	 
NI-DAQmx Runtime	National Instruments	Drivers	2023 Q2	 
NI-DAQmx Runtime with Configuration Support	National Instruments	Drivers	2023 Q2	 
NI-DAQmx Support for .NET Framework 4.0 Languages	National Instruments	Drivers	2023 Q2	 
NI-DAQmx Support for .NET Framework 4.5 Languages	National Instruments	Drivers	2023 Q2	 
NI-DAQmx Support for C	National Instruments	Drivers	2023 Q2	 
NI-VISA	National Instruments	Drivers	2022 Q3	 
NI-VISA .NET Development Support	National Instruments	Drivers	2022 Q3	 
NI-VISA .NET Runtime	National Instruments	Drivers	2022 Q3	 
NI-VISA C Examples	National Instruments	Drivers	2022 Q3	 
NI-VISA Configuration Support	National Instruments	Drivers	2022 Q3	 
NI-VISA Driver Development Wizard	National Instruments	Drivers	2022 Q3	 
NI-VISA Interactive Control	National Instruments	Drivers	2022 Q3	 
NI-VISA Runtime	National Instruments	Drivers	2022 Q3	 
PXI Platform Services Configuration	National Instruments	Drivers	2023 Q2	 
PXI Platform Services Runtime	National Instruments	Drivers	2023 Q2	 

[image]

## Issue:

When running the example ao read || write I received an error from DLL loading issue. The handle was None. 

Here is the code:
```
import nidaqmx
with nidaqmx.Task() as task:
    task.ai_channels.add_ai_voltage_chan("Dev1/ai0")
    task.read()
```

Here is the trace:
```
---------------------------------------------------------------------------
OSError                                   Traceback (most recent call last)
Cell In[6], line 2
      1 import nidaqmx
----> 2 with nidaqmx.Task() as task:
      3     task.ai_channels.add_ai_voltage_chan("Dev1/ai0")
      4     task.read()
[... 203 lines cut ...]
soupsieve                2.4
stack-data               0.6.2
streamlit                1.20.0
terminado                0.17.1
thonny                   4.0.2
tinycss2                 1.2.1
toml                     0.10.2
tomlkit                  0.11.6
toolz                    0.12.0
tornado                  6.2
traitlets                5.9.0
typing_extensions        4.5.0
tzdata                   2022.7
tzlocal                  4.3
uri-template             1.2.0
urllib3                  1.26.15
validators               0.20.0
watchdog                 3.0.0
wcwidth                  0.2.6
webcolors                1.13
webencodings             0.5.1
websocket-client         1.5.1
wheel                    0.38.4
wrapt                    1.15.0
zipp                     3.15.0

zhindes, 2023-04-20:
Assuming this is 64-bit Python, `nicaiu.dll` should be in "C:\Windows\System32". Do you have that file? If so, you can open it up in [Dependencies](https://github.com/lucasg/Dependencies) to see what dependency may be missing...

bkeryan, 2023-04-20:
`OSError: [WinError 1114] A dynamic link library (DLL) initialization routine failed` suggests that `nicaiu.dll` was found but one of the DLLs it depends on also failed to initialize.

Also, NTSTATUS 0xC000026C is `STATUS_DRIVER_UNABLE_TO_LOAD`. 
```
{Unable to Load Device Driver}
%hs device driver could not be loaded.
Error Status was 0x%x
...
C000026C
```

Is this a new DAQmx installation or an upgrade? Does the system have UEFI Secure Boot enabled?

zhindes, 2024-01-18:
No recent activity, and I suspect this is a lower-level driver install issue unrelated to nidaqmx-python. Closing.

closed 2024-01-18 by zhindes