ni/nidaqmx-python #341: latest NI-DAQmx Python API documentation is empty

matthias-tischler-ni, 2023-05-11:
when going to [[https://nidaqmx-python.readthedocs.io/]()](https://nidaqmx-python.readthedocs.io/) , you are automatically forwarded to https://nidaqmx-python.readthedocs.io/en/latest/

the individual pages like https://nidaqmx-python.readthedocs.io/en/latest/physical_channel.html are empty

nevertheless, the stable one is showing content, e.g. https://nidaqmx-python.readthedocs.io/en/stable/physical_channel.html

zhindes, 2023-05-11:
FYI @bkeryan - mainline documentation isn't working quite right. The latest release (0.7.0) is fine. [CONTRIBUTING.md](https://github.com/ni/nidaqmx-python/blob/master/CONTRIBUTING.md) details how to built it locally to check into it.

bkeryan, 2023-05-11:
I'll look into it. I wouldn't be surprised if https://github.com/ni/nidaqmx-python/pull/338 fixes it. It seems like we should add a doc generation check to tox.ini and possibly the PR build.

closed 2023-05-11 by bkeryan

bkeryan, 2023-05-11:
https://nidaqmx-python.readthedocs.io/en/latest/physical_channel.html has the correct content now.

PR #338 (2023-05-11): nidaqmx: Make gRPC optional
- [x] This contribution adheres to [CONTRIBUTING.md](https://github.com/ni/nidaqmx-python/blob/master/CONTRIBUTING.md).

### What does this Pull Request accomplish?

Make `nidaqmx` and a subset of its tests work without the `grpc` extra installed.

Update `tox.ini`:
- Test `nidaqmx` with and without the `grpc` extra.
- Test generating docs (without the `grpc` extra).

Also:
- Defer importing `_library_interpreter.py` or `_grpc_interpreter.py` until necessary
- Add some missing imports identified by mypy in my other branch
- Add type annotations for gRPC types to ver [... cut]