ni/nidaqmx-python #388: Acceleration channel yields unexpected results

aranvir, 2023-05-31:
## Summary
When measuring acceleration I get about +/-10g in idle (just noise). Measuring voltage yields +/- 0.04 mV. The certified sensor sensitivity is ~0.1 mV/g, so I would expect a noise level of +/- 0.4g. Any idea why the measured value is a bit more than 1 magnitude higher than it should be?

My hope is:
- It's a configuration issue on my side
- It's a python binding or an NI software issue

Of course, it could also be that the sensors are not certified/calibrated correctly or broken....... but since this is the worst case, I wanted to check in first if there are any software issues that I could look out for.

## Versions
Windows 10
Python 3.9.13
Nidaqmx 0.7.0
NI Version 2022 Q3

Acquisition unit: NI cDAQ-9174 with NI 9232
Sensor: PCB Piezotronics M350C02 Shock Accelerometer

## Details
First of all, this is the voltage measurement that I get from the NI Measurement and Automation explorer.

[image]

I can replicate this using `nidaqmx` with the following code:
```python
    def start_monitoring(self):
        self.buffer = np.zeros((self.num_sensors, self.channel_samples), dtype=np.float64)
        self.task = nidaqmx.Task()
        self.task.add_ai_voltage_chan("cDAQ1Mod1/ai0")
        self.task.timing.cfg_samp_clk_timing(102e3, sample_mode=constants.AcquisitionType.CONTINUOUS,
                                             samps_per_chan=50000)
        reader = stream_readers.AnalogMultiChannelReader(self.task.in_stream)

        def reading_task_callback(task_idx, event_type, num_samples, callback_data=None):  # pylint: disable=W0613
            reader.read_many_sample(self.buffer, num_samples, timeout=constants.WAIT_INFINITELY)
            self.buffer_read.emit(self.buffer)  # This is a Qt signal that I use to forward the reading to a buffer used for plotting
            return 0

        self.task.register_every_n_samples_acquired_into_buffer_event(self.channel_samples, reading_task_callback)
        logging.debug("Starting task")
        self.task.start()
```

[image]

Same max. amplitude, same pattern, so looks good to me. Now if I switch out `add_ai_voltage_chan` for `add_ai_accel_chan` as follows
```python
        self.task.add_ai_accel_chan(
            "cDAQ1Mod1/ai0",
            units=AccelUnits.G,
            min_val=-10000,
            max_val=10000,
            sensitivity=0.1,
            sensitivity_units=AccelSensitivityUnits.MILLIVOLTS_PER_G
)
```
I get the following results. Much denser plotter, and max amplitude of about 10g.
[image]

So yea, I would have expected a plot similar to the mV one with a max amplitude of about 0.4g. Any ideas?

aranvir, 2023-05-31:
Update: I also tested it with some other sensor (fresh delivery) and another DAQ. Same results. So, I don't think this is a hardware issue.

zhindes, 2023-05-31:
@aranvir - With a sensitivity of 0.1 mV/g, I would expect your final graph to be +/-400g, right? The voltage graph is +/- 40mV, divide by the sensitivity, and you get g. That being said your final graph still looks wrong to me. What happens if you change to this code that should be functionally equivalent?

```python
self.task.add_ai_accel_chan(
   "cDAQ1Mod1/ai0",
   units=AccelUnits.G,
   min_val=-10000,
   max_val=10000,
   sensitivity=0.0001,
   sensitivity_units=AccelSensitivityUnits.VOLTS_PER_G
)
```

zhindes, 2023-05-31:
@aranvir - I played around with this, and I am getting correct scaling in all the various configuration I have tried. I did notice that there are three other optional parameters to `add_ai_accel_chan` that you aren't providing.

```python
current_excit_source=ExcitationSource.INTERNAL,
current_excit_val=0.004,
custom_scale_name=""):
```

`custom_scale_name` is fine, but the other two are important. I suspect that is the major difference between voltage and acceleration measurements in your example. The test panel does have an `IEPE Enable` button that you can check that might produce a similar waveform to what you're seeing in the Acceleration graph.

aranvir, 2023-06-01:
@zhindes thanks for the feedback!

You are right, the volt reading is 0.04 V or 40 mV.... tunnelvision. So yea, the expected acceleration is then 400g. Which seems high for no excitation.. but then again, maybe just basic noise level for a sensor that's supposed to measure in a +/- 10.000g range.

Yep, I kept the default values for the other 3 parameters. I will look into this!

zhindes, 2023-06-01:
Let us know if anything else comes up :)

aranvir, 2023-06-06:
Hi @zhindes , I played around with the values in the explorer and when I set it to AC and enable IEPE, I get the result below. Now the shape of the graphs match much better. Also, the expected acceleration is now only around 8g which is much closer to what we are actually measuring.

[image]

But it seems the sensitivity is a bit off since the max. amplitude is closer to 10g than 8g. Still, much better than I feared based on the initial values before.

[image]

I also played with the excitation current, but that did not alter the result notably. So I'm just gonna leave it at the default value.

zhindes, 2023-06-09:
@aranvir - A couple of differences (maybe) between the Test Panel and your Python application
* *Terminal Configuration* - I think for acceleration, `TerminalConfiguration.DEFAULT` is the same as `TerminalConfiguration.PSEUDO_DIFF`, but you choose that explicitly to be sure.
* *Coupling* - Coupling isn't in the create channel for `add_ai_accel_chan`. You can set it explicitly like this:

```python
chan.ai_coupling = Coupling.AC
```

Also, your **sensitivity** value of 0.1 mV/G sounds like a nominal value. Your specific sensor may have a calibration certificate that has a more exact value. It should be very close to that, though.